openXC7/nextpnr-xilinx #11: Error messaging does not provide enough info for the self-guided debug

chili-chips-ba, 2023-08-17:
"**Failed assert**" was reported for an issue that had nothing to do with assertions.
[https://symbioticeda.slack.com/archives/C053XV4R248/p1691040074754449](url)

We also strongly suggest adding switch for control of tool messaging verbosity level. This is even more important given the very limited openXC7 user base, hence no useful finds in Internet searches.

hansfbaier, 2024-10-13:
@chili-chips-ba This link seems to be broken/lead to nothing anymore.
I know that the error messaging is not very good,
and I improved some. Do you still have concrete issues with this?
What are they?

chili-chips-ba, 2024-10-14:
[image]

[image]

hansfbaier, 2024-10-14:
Ok that one already has been fixed

closed 2024-10-14 by chili-chips-ba