openXC7/nextpnr-xilinx #15: XDC Parsing fails without human-readable error

gh-sh4, 2023-10-21:
Running nextpnr-xilinx version `Version 0.6.0-1-g55beb766` for part `xc7a200tsbg484-1`

I'm trying to file an issue for nextpnr-xilinx failing to handle my yosys output. This is nearly unchanged from the official XDC file from Digilent for the Nexys Video. Along the way, I found out the xdc file itself even fails parsing, but there is no error message.

I am running via the command line
```
./nextpnr-xilinx --chipdb ~/nexys.bin --json ~/Dragon.json --xdc ~/maybe.xdc
```

## Parser overly (?) sensitive to whitespace
I am not sure exactly what the case is since I had to bulk edit many lines several times to get it to pass this stage, but some part of the XDC parser is very sensitive to spaces, e.g. `} ]` vs `}]`. I can dig into this more, but ideally if there is a parse error, it does not abort but will report the line which fails to the user (and then maybe abort if no more useful message can be made).

# General failure to parse XDC
In the attached file I have some other parse error which has no human-readable error. There is again no human-readable error to help figure out what is wrong. Running in gdb via `gdb --args ./nextpnr-xilinx --chipdb ~/nexys.bin --json ~/Dragon.json --xdc ~/maybe.xdc` leads to 

```
terminate called after throwing an instance of 'std::out_of_range'
  what():  dict::at()

Program received signal SIGABRT, Aborted.
__pthread_kill_implementation (no_tid=0, signo=6, threadid=140737351878464) at ./nptl/pthread_kill.c:44
44      ./nptl/pthread_kill.c: No such file or directory.
(gdb) bt
#0  __pthread_kill_implementation (no_tid=0, signo=6, threadid=140737351878464) at ./nptl/pthread_kill.c:44
#1  __pthread_kill_internal (signo=6, threadid=140737351878464) at ./nptl/pthread_kill.c:78
#2  __GI___pthread_kill (threadid=140737351878464, signo=signo@entry=6) at ./nptl/pthread_kill.c:89
#3  0x00007ffff7042476 in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#4  0x00007ffff70287f3 in __GI_abort () at ./stdlib/abort.c:79
#5  0x00007ffff74a2b9e in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#6  0x00007ffff74ae20c in ?? () from /lib/x86_64-linux-gnu/libstdc++.so.6
#7  0x00007ffff74ae277 in std::terminate() () from /lib/x86_64-linux-gnu/libstdc++.so.6
#8  0x00007ffff74ae4d8 in __cxa_throw () from /lib/x86_64-linux-gnu/libstdc++.so.6
#9  0x00005555555dc6d7 in nextpnr_xilinx::dict<nextpnr_xilinx::IdString, nextpnr_xilinx::IdString, nextpnr_xilinx::hash_ops<nextpnr_xilinx::IdString> >::at (this=0x555555bb8a70, key=...)
    at /home/sh4/p/nextpnr-xilinx/common/kernel/hashlib.h:597
#10 0x00005555555d918f in nextpnr_xilinx::BaseCtx::getNetByAlias (this=0x555555bb7c90, alias=...) at /home/sh4/p/nextpnr-xilinx/common/kernel/basectx.h:216
#11 0x000055555598837e in operator() (__closure=0x7fffffffd1f0, str="get_ports {clk}") at /home/sh4/p/nextpnr-xilinx/xilinx/xdc.cc:110
#12 0x0000555555988d1a in nextpnr_xilinx::Arch::parseXdc (this=0x555555bb7c90, in=...) at /home/sh4/p/nextpnr-xilinx/xilinx/xdc.cc:172
#13 0x0000555555927631 in UspCommandHandler::customAfterLoad (this=0x7fffffffd950, ctx=0x555555bb7c90) at /home/sh4/p/nextpnr-xilinx/xilinx/main.cc:83
#14 0x0000555555604ad7 in nextpnr_xilinx::CommandHandler::executeMain (this=0x7fffffffd950, ctx=std::unique_ptr<nextpnr_xilinx::Context> = {...}) at /home/sh4/p/nextpnr-xilinx/common/kernel/command.cc:431
#15 0x0000555555606e00 in nextpnr_xilinx::CommandHandler::exec (this=0x7fffffffd950) at /home/sh4/p/nextpnr-xilinx/common/kernel/command.cc:549
#16 0x00005555559277a7 in main (argc=7, argv=0x7fffffffddf8) at /home/sh4/p/nextpnr-xilinx/xilinx/main.cc:91
```

so I assume there is some issue finding a particular net. Debugging this is a little difficult since the net names are stored in a dictionary (which is good, but makes seeing which net this is harder).

The general ask is to add error messages before aborting.

[Dragon.tar.gz](https://github.com/openXC7/nextpnr-xilinx/files/13061706/Dragon.tar.gz)
[maybe.xdc.txt](https://github.com/openXC7/nextpnr-xilinx/files/13061679/maybe.xdc.txt)

hansemro, 2023-10-29:
Yes, I can confirm there is a bug with the parser when there are spaces inside curly braces.

Looking at openXC7/demo-projects/blinky-kc705/blinky.xdc for example, the following lines are acceptable with nextpnr's xdc parser:
- `set_property LOC AD12 [get_ports clk_p]`
- `set_property LOC AD12 [get_ports {clk_p}]`

However, the following will throw assertion errors (without useful error messages):
- `set_property LOC AD12 [get_ports { clk_p }]`
- `set_property LOC AD12 [get_ports { clk_p}]`
- `set_property LOC AD12 [get_ports {clk_p }]`

Assertion error:
```
nextpnr-xilinx --chipdb ../chipdb//xc7k325tffg900.bin --xdc blinky.xdc --json blinky.json --fasm blinky.fasm  
terminate called after throwing an instance of 'nextpnr_xilinx::assertion_failure'
  what():  Assertion failure: str.back() == '}' (/build/source/xilinx/xdc.cc:41)
make: *** [../openXC7.mk:39: blinky.fasm] Aborted (core dumped)
```

hansfbaier, 2023-11-04:
Yes the capability of the XDC parser is quite limited. I would avoid dicts for now

hansfbaier, 2023-11-04:
@hansemro Thanks for the observations. That will help fixing it.

gh-sh4, 2023-11-07:
@hansfbaier  I'm happy to make some PRs to improve the situation here, at the very least adding some logs or messages to asserts. Would you be alright reviewing those if I make the PRs (at whatever priority makes sense to you)?

hansfbaier, 2023-11-07:
Yes sure, thanks for the offer.

gh-sh4, 2023-12-05:
Feel free to reopen if you like, but I believe all exceptions during XDC parsing will now at least fall into a catch-all block and log the line which is unhandled. This is already a big improvement. I'm closing this issue as resolved.

closed 2023-12-05 by gh-sh4