openXC7/nextpnr-xilinx #17: Crash during DSP check_illegal_fanout

gh-sh4, 2023-11-09:
I am currently trying to synthesize a project (same constraints and project file given in my previous issue (#15). 

When invoking `./build/nextpnr-xilinx --chipdb nexys.bin --json ~/p/dragon/build-xc7/hdl/Dragon.json --xdc ~/p/dragon/board_support/nexys/Nexys-Video-Master.xdc` there is an assertion which triggers, and I cannot understand how the code here is intended to work. Hoping we can make some progress and get a fix in (I can potentially help once I know what is supposed to happen in this code).

Final lines before debug assert:
```
...
Info: Packing BRAM..
Info: Packing DSPs..
Info:     Created 16 DSP48E1_DSP48E1 cells from:
Info:             16x DSP48E1
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 1) >= this->size() (which is 1)
```
GDB Backtrace
```
(gdb) bt
#0  0x00007ffff72b0884 in __pthread_kill_implementation () from /lib64/libc.so.6
#1  0x00007ffff725fafe in raise () from /lib64/libc.so.6
#2  0x00007ffff724887f in abort () from /lib64/libc.so.6
#3  0x00007ffff74a4d19 in __gnu_cxx::__verbose_terminate_handler() [clone .cold] () from /lib64/libstdc++.so.6
#4  0x00007ffff74b4f4c in __cxxabiv1::__terminate(void (*)()) () from /lib64/libstdc++.so.6
#5  0x00007ffff74b4fb7 in std::terminate() () from /lib64/libstdc++.so.6
#6  0x00007ffff74b5218 in __cxa_throw () from /lib64/libstdc++.so.6
#7  0x00007ffff74a7765 in std::__throw_out_of_range_fmt(char const*, ...) [clone .cold] () from /lib64/libstdc++.so.6
#8  0x00000000004c4ebf in std::vector<nextpnr_xilinx::indexed_store<nextpnr_xilinx::PortRef>::slot, std::allocator<nextpnr_xilinx::indexed_store<nextpnr_xilinx::PortRef>::slot> >::_M_range_check (this=0xbca82c0, __n=1) at /usr/include/c++/13/bits/stl_vector.h:1155
#9  0x00000000004c4327 in std::vector<nextpnr_xilinx::indexed_store<nextpnr_xilinx::PortRef>::slot, std::allocator<nextpnr_xilinx::indexed_store<nextpnr_xilinx::PortRef>::slot> >::at (this=0xbca82c0, __n=1) at /usr/include/c++/13/bits/stl_vector.h:1177
#10 0x00000000004c3d5c in nextpnr_xilinx::indexed_store<nextpnr_xilinx::PortRef>::at (this=0xbca82c0, idx=...)
    at /home/sh4/p/nextpnr-xilinx/common/kernel/indexed_store.h:165
#11 0x00000000004c38be in nextpnr_xilinx::indexed_store<nextpnr_xilinx::PortRef>::iterator::operator* (this=0x7fffffffc450)
    at /home/sh4/p/nextpnr-xilinx/common/kernel/indexed_store.h:202
#12 0x00000000007b90fb in operator() (__closure=0x7fffffffc4f0, ni=0xbca82a0, port="PCOUT47") at /home/sh4/p/nextpnr-xilinx/xilinx/pack_dsp_xc7.cc:34
#13 0x00000000007b9325 in nextpnr_xilinx::XC7Packer::walk_dsp (this=0x7fffffffc900, root=0xbca9c10, current_cell=0xbca9c10, constr_z=25)
    at /home/sh4/p/nextpnr-xilinx/xilinx/pack_dsp_xc7.cc:47
#14 0x00000000007b9f8f in nextpnr_xilinx::XC7Packer::pack_dsps (this=0x7fffffffc900) at /home/sh4/p/nextpnr-xilinx/xilinx/pack_dsp_xc7.cc:158
#15 0x0000000000790a34 in nextpnr_xilinx::Arch::pack (this=0x9e8af0) at /home/sh4/p/nextpnr-xilinx/xilinx/pack.cc:905
#16 0x000000000049a1c1 in nextpnr_xilinx::CommandHandler::executeMain (this=0x7fffffffd6d0, ctx=std::unique_ptr<nextpnr_xilinx::Context> = {...})
    at /home/sh4/p/nextpnr-xilinx/common/kernel/command.cc:452
#17 0x000000000049bd47 in nextpnr_xilinx::CommandHandler::exec (this=0x7fffffffd6d0) at /home/sh4/p/nextpnr-xilinx/common/kernel/command.cc:549
#18 0x0000000000784cdf in main (argc=7, argv=0x7fffffffdb78) at /home/sh4/p/nextpnr-xilinx/xilinx/main.cc:91
```

If we look here in the code @ pack_dsp_xc7.cc
```cpp
    auto check_illegal_fanout = [&] (NetInfo *ni, std::string port) {
        if (ni->users.entries() > 1)
            log_error("Port %s connected to net %s has more than one user", port.c_str(), ni->name.c_str(ctx));

34:        PortRef& user = *ni->users.end();
        if (user.cell->type != id_DSP48E1_DSP48E1)
            log_error("User %s of net %s is not a DSP block, but %s",
                user.cell->name.c_str(ctx), ni->name.c_str(ctx), user.cell->type.c_str(ctx));
    };
```
Line 34 is immediately dereferencing the end iterator of a container. In any standard library or other custom container I've ever seen, end represents an exclusive end (i.e. "the locastion after the last valid entry"), so this would appear on the surface to be just broken. Do you understand what is intended here? Perhaps I'm reading it wrong.

hansfbaier, 2023-11-09:
I think it means the last user in the collection.

hansfbaier, 2023-11-09:
I need to mention that using the main branch is currently not recommended for productive work, because
it is quite broken. The stable-backports branch is the way to go.

hansfbaier, 2023-11-09:
And there we have back instead of end
[image]

gh-sh4, 2023-11-09:
> And there we have back instead of end [image]

Yes, I did a bit of digging and this problem is in a few places.

hansfbaier, 2023-11-09:
We have to look into indexed_store.h
for a suitable method

hansfbaier, 2023-11-09:
You could try begin(). That should work.

hansfbaier, 2023-11-09:
I just pushed this fix to main. Let me know if that fixes it.

closed 2023-11-09 by hansfbaier

gh-sh4, 2023-11-09:
> You could try begin(). That should work.

Yes, I've done that and have a new failure later. Would you like another issue? Or just continue here.

hansfbaier, 2023-11-09:
Can I see it?

hansfbaier, 2023-11-09:
Make a new one, I guess. Let's close this.

hansfbaier, 2023-11-09:
But I have to repeat: The main branch is fundamentally broken in the moment.
It is not recommended for building your designs.
Use the stable-backports branch.

gh-sh4, 2023-11-10:
I'll create a separate issue for the next fail as you requested. I commented on your commit (https://github.com/openXC7/nextpnr-xilinx/commit/27176b921d8420845f5069d0a3836b2120d7a970), as this same issue will also fail at another line without further changes.