openXC7/nextpnr-xilinx #20: Limited support for Distributed Memory / LUTRAM

hansemro, 2023-12-20:
`memory_libmap` pass in Yosys 0.18 and newer exposed support deficiencies in nextpnr-xilinx for single-port LUTRAMs including:
- RAMS32
- RAMS64E
- RAM32X1S
- RAM64X1S
- RAM64X1S_1
- RAM128X1S
- RAM256X1S

Most dual-port LUTRAMs (one read address, one write address) should already be handled by nextpnr-xilinx (TODO: validate). These include:
- RAMD32
- RAMD64E
- RAM32X1D
- RAM64X1D
- RAM128X1D
- RAM256X1D (for ultrascale)

Whole-slice LUTRAMs (instances that occupy all LUT BELs in a site) should also be supported (TODO: verify). These include:
- RAM32M
- RAM64M

## Issue Description

TODO: describe issue

## Tasks/Status

- [ ] XilinxPacker::pack_dram transforms all remaining LUTRAMs (mostly single-port):
    - [ ] define transform rules for RAMS32
    - [ ] define transform rules for RAMS64E
    - [ ] transform RAM32X1S to RAMS32
        - [ ] update create_dram32_lut to handle single-port
    - [ ] transform RAM64X1S to RAMS64E
        - [ ] update create_dram_lut to handle single-port
    - [ ] transform RAM128X1S
    - [ ] transform RAM256X1S
- [ ] update fasm
    - [ ] handle RAMS32
    - [ ] handle RAMS64E
- [ ] validate

## Development Branches

- experimental branch with support for RAMS32, RAMS64E, RAM32X1S, RAM64X1S: https://github.com/hansemro/nextpnr-xilinx/commits/xc7-lutram-dev/
    - use with yosys 0.18 or newer to test
    - rebuild chipdb after building
    - currently untested on live FPGA

hansfbaier, 2023-12-20:
Very good, thanks!

hansemro, 2023-12-20:
Added support for RAM128X1S and RAM256X1S though not sufficiently tested. I was able to build litex-ddr-kc705 after latest changes. However, I am now running into DDR memtest issues it seems: https://gist.github.com/hansemro/5f48f4098e59f9db2e34ae25cb0b6ecd

hansfbaier, 2023-12-20:
@hansemro Wow, that was quick! We might want to write some basic tests here:
https://github.com/openXC7/primitive-tests
Debugging the issue inside that complex design is probably too cumbersome.

hansemro, 2023-12-20:
@hansfbaier Sounds good. I'll try to write some tests soon. DDR issue could be totally unrelated to this.

hansfbaier, 2023-12-20:
@hansemro Yes, very likely your code works. I never got 8 modules working nice with OpenXC7. At some point the timing just falls apart, because of congestion, I suppose. See my comment on your gist.

hansemro, 2023-12-20:
Rebased experimental branch on https://github.com/openXC7/nextpnr-xilinx/commit/8120acd7c036cd87af2f1169e9c555546ad7bded (current stable-backports)

hansfbaier, 2023-12-20:
@hansemro great, thanks

hansemro, 2023-12-20:
Fixed up RAM32X1D not creating RAMD32 instances and made sure RAM32X1S is handled in pack_dram (forgot this one apparently).

hansemro, 2023-12-20:
Made initial tests: https://github.com/hansemro/primitive-tests/commits/lutram-tests/

Notably nextpnr-xilinx hangs with RAM256X1S. Seems #10 made this observation months ago. I'll try to spend some time debugging this.

hansemro, 2023-12-20:
RAM256X1S and RAM256X1D were not handled correctly since their address pins are in an array A[N:0] rather than specified individually (A0, A1, A2, ... ). I'll need to validate every transform rule anyway...

hansfbaier, 2023-12-20:
@hansemro I pushed an MMCM-fix to stable-backports. The MMCM should work now, if you have time to try.

hansemro, 2023-12-21:
@hansfbaier Thanks for the heads up. I will get to it at some point. I decided it is better for me to fix dual-port LUTRAMs before moving forward, because everything is broken (for at least xc7, not sure about ultrascale).

hansfbaier, 2023-12-21:
Yes definitely, that is more important. Thanks for working on this.

hansemro, 2023-12-21:
I misspoke since I was testing RAM256X1D (thought I was testing RAM256X1S) which does not fit in xc7 anyhow. Still, yosys should not allow unsupported LUTRAMs to be synthesized!

hansemro, 2023-12-21:
Resolved two issues:
- Fixed address port detection for single port LUTRAM with c073116536800022adbdeb0ba4b785026a6116b4
- Fixed MUXF tree z offset for RAM256X1S which broke placer with 4792453ba7bde361d090e91a18456a1601e7b5e2

hansemro, 2023-12-21:
> I pushed an MMCM-fix to stable-backports. The MMCM should work now, if you have time to try.

Still not working on KC705 (xc7k325tffg900).

```
CMT_TOP_L_LOWER_B_X228Y113.MMCME2_ADV.IN_USE
CMT_TOP_L_LOWER_B_X228Y113.MMCME2_ADV.ZINV_PWRDWN
CMT_TOP_L_LOWER_B_X228Y113.MMCME2_ADV.ZINV_PSEN
CMT_TOP_L_LOWER_B_X228Y113.MMCME2_ADV.ZINV_PSINCDEC
CMT_TOP_L_LOWER_B_X228Y113.MMCME2_ADV.DIVCLK_DIVCLK_HIGH_TIME[5:0] = 6'b000001
CMT_TOP_L_LOWER_B_X228Y113.MMCME2_ADV.DIVCLK_DIVCLK_LOW_TIME[5:0] = 6'b000001
CMT_TOP_L_LOWER_B_X228Y113.MMCME2_ADV.DIVCLK_DIVCLK_NO_COUNT[0]
CMT_TOP_L_LOWER_B_X228Y113.MMCME2_ADV.CLKFBOUT_CLKOUT1_OUTPUT_ENABLE[0]
CMT_TOP_L_LOWER_B_X228Y113.MMCME2_ADV.CLKFBOUT_CLKOUT1_HIGH_TIME[5:0] = 6'b000011
CMT_TOP_L_LOWER_B_X228Y113.MMCME2_ADV.CLKFBOUT_CLKOUT1_LOW_TIME[5:0] = 6'b000011
CMT_TOP_L_LOWER_B_X228Y113.MMCME2_ADV.CLKFBOUT_CLKOUT1_PHASE_MUX[2:0] = 3'b000
CMT_TOP_L_LOWER_B_X228Y113.MMCME2_ADV.CLKFBOUT_CLKOUT2_DELAY_TIME[5:0] = 6'b000000
CMT_TOP_L_LOWER_B_X228Y113.MMCME2_ADV.CLKOUT0_CLKOUT1_OUTPUT_ENABLE[0]
CMT_TOP_L_LOWER_B_X228Y113.MMCME2_ADV.CLKOUT0_CLKOUT1_HIGH_TIME[5:0] = 6'b001000
CMT_TOP_L_LOWER_B_X228Y113.MMCME2_ADV.CLKOUT0_CLKOUT1_LOW_TIME[5:0] = 6'b001000
CMT_TOP_L_LOWER_B_X228Y113.MMCME2_ADV.CLKOUT0_CLKOUT1_PHASE_MUX[2:0] = 3'b000
CMT_TOP_L_LOWER_B_X228Y113.MMCME2_ADV.CLKOUT0_CLKOUT2_DELAY_TIME[5:0] = 6'b000000
CMT_TOP_L_LOWER_B_X228Y113.MMCME2_ADV.CLKOUT1_CLKOUT1_OUTPUT_ENABLE[0]
CMT_TOP_L_LOWER_B_X228Y113.MMCME2_ADV.CLKOUT1_CLKOUT1_HIGH_TIME[5:0] = 6'b000010
CMT_TOP_L_LOWER_B_X228Y113.MMCME2_ADV.CLKOUT1_CLKOUT1_LOW_TIME[5:0] = 6'b000010
CMT_TOP_L_LOWER_B_X228Y113.MMCME2_ADV.CLKOUT1_CLKOUT1_PHASE_MUX[2:0] = 3'b000
CMT_TOP_L_LOWER_B_X228Y113.MMCME2_ADV.CLKOUT1_CLKOUT2_DELAY_TIME[5:0] = 6'b000000
CMT_TOP_L_LOWER_B_X228Y113.MMCME2_ADV.CLKOUT2_CLKOUT1_OUTPUT_ENABLE[0]
CMT_TOP_L_LOWER_B_X228Y113.MMCME2_ADV.CLKOUT2_CLKOUT1_HIGH_TIME[5:0] = 6'b000011
CMT_TOP_L_LOWER_B_X228Y113.MMCME2_ADV.CLKOUT2_CLKOUT1_LOW_TIME[5:0] = 6'b000011
CMT_TOP_L_LOWER_B_X228Y113.MMCME2_ADV.CLKOUT2_CLKOUT1_PHASE_MUX[2:0] = 3'b000
CMT_TOP_L_LOWER_B_X228Y113.MMCME2_ADV.CLKOUT2_CLKOUT2_DELAY_TIME[5:0] = 6'b000000
CMT_TOP_L_LOWER_B_X228Y113.MMCME2_ADV.COMP.Z_ZHOLD
CMT_TOP_L_LOWER_B_X228Y113.MMCME2_ADV.FILTREG1_RESERVED[11:0] = 12'b000000001000
CMT_TOP_L_LOWER_B_X228Y113.MMCME2_ADV.LKTABLE[39:0] = 40'b1110011100111110100011111010010000000001
CMT_TOP_L_LOWER_B_X228Y113.MMCME2_ADV.POWER_REG_POWER_REG_POWER_REG[15:0] = 16'b1001100100000000
CMT_TOP_L_LOWER_B_X228Y113.MMCME2_ADV.LOCKREG3_RESERVED[0]
CMT_TOP_L_LOWER_B_X228Y113.MMCME2_ADV.TABLE[9:0] = 10'b1111010100
```

hansfbaier, 2024-01-15:
@hansemro How are things going? Have you been able to test the changes?

hansemro, 2024-02-06:
@hansfbaier

> Have you been able to test the changes?

MMCM is confirmed working with multiple clock outputs on KC705 though with a BUFG on all clock outputs. fasm2frames would throw segment DB errors if I didn't have them. Test branch: https://github.com/hansemro/primitive-tests/commits/mmcm-blinky-kc705-db-error/

Interestingly, the first clock output didn't require me to place BUFG buffer, though it should probably have one.

> How are things going?

Things were going well until I had to handle each LUTRAM as an edge case. Initially, I was less bothered to write things down, but now I feel it is appropriate to actually spend time documenting the port/parameter transformations for all cells (including ultrascale-only cells). I intend to resume work on validation and get xc7 cells covered.

Anyhow, it turns out I made some incorrect assumptions about some things that I will need to handle:

1. create_dram32_lut does not currently support mapping RAMS32 to LUT5 BEL site
2. check whether address ports are connected to both A{6:1} and WA{8:1} (WA{9:1} for ultrascale) ports in LUT6/LUT5 BEL for single-port LUTRAM

Will elaborate further with a follow-up post on LUTRAM transformations to RAMS/RAMD primitive cells to LUT5/LUT6 BELs in more detail.

hansfbaier, 2024-02-06:
Yes, I had similar observations. 
CLKOUT 1-3 had missing pips. CLKOUT 0,5,6 worked fine out of the box:
https://github.com/openXC7/primitive-tests/blob/main/mmcm-blinky-kintex/blinky.v

hansfbaier, 2024-02-06:
Thanks for the update!

hansemro, 2024-02-08:
## Naming Notation:

I'll define the following name notation to simplify/reduce port/parameter names:

- A comma-separated list or range enclosed by curly brackets {} maps to individual names with an entry from the list/range.
    - Curly brackets may be inserted anywhere in the name.
    - List example: `ADR{3, 2, 1, 0}` maps to `ADR3, ADR2, ADR1, ADR0`
    - Range example: `{D:A}_O` maps to `D_O, C_O, B_O, A_O`
- Verilog-style array with a range enclosed by square brackets [].
    - The range indicates the bit width of the signal/parameter. If there are no square brackets, assume 1-bit width.
    - Square brackets can only be placed at the end of a name.
    - Example: `ADR[3:0]` maps to `ADR[3], ADR[2], ADR[1], ADR[0]`
- Combined example:
    - `ADDR{D:A}[4:0]` maps to `ADDRD[4:0], ADDRC[4:0], ADDRB[4:0], ADDRA[4:0]`

## LUTRAM Cell Table:

Additional notes:
- LUT_OR_MEM5/LUT_OR_MEM6 BEL cannot be instantiated manually. Port and parameter details of these BELs are purely behavioral.
- LUT_OR_MEM6 BEL has a MC31 pin which is used for a shift register and can be ignored for LUTRAM.

| Cell | Cell Type | US-only | `*INIT*` Parameter | `*CLK_INVERTED` Parameter | Clock Input | Write Enable Input | Write Data Input | Write Select Input | Write Address Input | Read Address Input | Read Data Output |
| ---- | ---- | ---- | ---- | ---- | ---- | ---- | ---- | ---- | ---- | ---- | ---- |
| LUT_OR_MEM5 | BEL | false | `INIT[31:0]` | N/A | `CLK` | `WE` | `DI1` | N/A | `WA{5:1}` | `A{5:1}` | `O5` |
| LUT_OR_MEM6 | BEL | false | `INIT[63:0]` | N/A | `CLK` | `WE` | `DI2` | N/A | US ? `WA{9:1}` : `WA{8:1}` | `A{6:1}` | `O6` |
| RAMS32 |  LUTRAM Primitive | false | `INIT[31:0]` | `IS_CLK_INVERTED` | `CLK` | `WE` | `I` | N/A | `ADR{4:0}` | `ADR{4:0}` | `O` |
| RAMD32 |  LUTRAM Primitive | false | `INIT[31:0]` | `IS_CLK_INVERTED` | `CLK` | `WE` | `I` | N/A | `WADR{4:0}` | `RADR{4:0}` | `O` |
| RAMD32M64 |  LUTRAM Primitive | true | `INIT[63:0]` | `IS_CLK_INVERTED` | `CLK` | `WE` | `I` | N/A | `WADR{4:0}` | `RADR{5:0}` | `O` |
| RAM32X1S | LUTRAM | false | `INIT[31:0]` | `IS_WCLK_INVERTED` | `WCLK` | `WE` | `D` | N/A | `A{4:0}` | `A{4:0}` | `O` |
| RAM32X1D | LUTRAM | false | `INIT[31:0]` | `IS_WCLK_INVERTED` | `WCLK` | `WE` | `D` | N/A | `A{4:0}` | `A{4:0}`; `DPRA{4:0}` | `SPO`; `DPO` |
| RAM32X16DR8 | Asymmetric LUTRAM | true | N/A? | `IS_WCLK_INVERTED` | `WCLK` | `WE` | `DI{H:A}[1:0]` | N/A | `ADDRH[4:0]`;`ADDR{G:A}[5:0]` | `ADDRH[4:0]`;`ADDR{G:A}[5:0]` | `DOH[1:0]`; `DO{G:A}` |
| RAM32M | SelectRAM | false | `INIT_{D:A}[63:0]` | `IS_WCLK_INVERTED` | `WCLK` | `WE` | `DI{D:A}[1:0]` | N/A | `ADDR{D:A}[4:0]` | `ADDR{D:A}[4:0]` | `DO{D:A}[1:0]` |
| RAM32M16 | SelectRAM | true | `INIT_{H:A}[63:0]` | `IS_WCLK_INVERTED` | `WCLK` | `WE` | `DI{H:A}[1:0]` | N/A | `ADDR{H:A}[4:0]` | `ADDR{H:A}[4:0]` | `DO{H:A}[1:0]` |
| RAMS64E |  LUTRAM Primitive | false | `INIT[63:0]` | `IS_CLK_INVERTED` | `CLK` | `WE` | `I` | N/A | (`WADR{7:6}`), `ADR{5:0}` | `ADR{5:0}` | `O` |
| RAMS64E1 |  LUTRAM Primitive | true? | `INIT[63:0]` | `IS_CLK_INVERTED` | `CLK` | `WE` | `I` | N/A | (`WADR{8:6}`), `ADR{5:0}` | `ADR{5:0}` | `O` |
| RAMD64E |  LUTRAM Primitive | false | `INIT[63:0]` | `IS_CLK_INVERTED` | `CLK` | `WE` | `I` | N/A | `WADR{7:0}` | `RADR{5:0}` | `O` |
| RAM64X1S | LUTRAM | false | `INIT[63:0]` | `IS_WCLK_INVERTED` | `WCLK` | `WE` | `D` | N/A | `A{5:0}` | `A{5:0}` | `O` |
| RAM64X1D | LUTRAM | false | `INIT[63:0]` | `IS_WCLK_INVERTED` | `WCLK` | `WE` | `D` | N/A | `A{5:0}` | `A{5:0}`; `DPRA{5:0}` | `SPO`; `DPO` |
| RAM64X8SW | SelectRAM | true | `INIT_{H:A}[63:0]` | `IS_WCLK_INVERTED` | `WCLK` | `WE` | `D` | `WSEL[2;0]` | `A[5;0]` | `A[5;0]` | `O[7:0]` |
| RAM64M | SelectRAM | false | `INIT_{D:A}[63:0]` | `IS_WCLK_INVERTED` | `WCLK` | `WE` | `DI{D:A}` | N/A | `ADDR{D:A}[5:0]` | `ADDR{D:A}[5:0]` | `DO{D:A}` |
| RAM64M8 | SelectRAM | true | `INIT_{H:A}[63:0]` | `IS_WCLK_INVERTED` | `WCLK` | `WE` | `DI{H:A}` | N/A | `ADDR{H:A}[5:0]` | `ADDR{H:A}[5:0]` | `DO{H:A}` |
| RAM128X1S | LUTRAM | false | `INIT[127:0]` | `IS_WCLK_INVERTED` | `WCLK` | `WE` | `D` | N/A | `A{6:0}` | `A{6:0}` | `O` |
| RAM128X1D | LUTRAM | false | `INIT[127:0]` | `IS_WCLK_INVERTED` | `WCLK` | `WE` | `D` | N/A | `A[6:0]` | `A[6:0]`; `DPRA[6:0]` | `SPO`; `DPO` |
| RAM256X1S | LUTRAM | false | `INIT[255:0]` | `IS_WCLK_INVERTED` | `WCLK` | `WE` | `D` | N/A | `A[7:0]` | `A[7:0]` | `O` |
| RAM256X1D | LUTRAM | true | `INIT[255:0]` | `IS_WCLK_INVERTED` | `WCLK` | `WE` | `D` | N/A | `A[7:0]` | `A[7:0]`; `DPRA[7:0]` | `SPO`; `DPO` |
| RAM512X1S | LUTRAM | true | `INIT[511:0]` | `IS_WCLK_INVERTED` | `WCLK` | `WE` | `D` | N/A | `A[8:0]` | `A[8:0]` | `O` |

hansemro, 2024-02-08:
## XC7 LUTRAM to LUTRAM Primitive Transformations:

LUTRAMs are broken down to primitive cell(s) that will eventually map to SLICEM LUT_OR_MEM6/LUT_OR_MEM5 BEL site(s) once placed.

Convention:
- Title: Source Cell to Destination Cell
- Transformations: Source Name to Destination Name (if owner is not specified)

### RAM32X1S -> 1x RAMS32

Cell Rules:
- RAMS32 (6LUT) with `/SP` appended to name
- No MUXF cells
- 1 output

Parameter Rules:
- `IS_WCLK_INVERTED` to `IS_CLK_INVERTED`

Port Rules:
- `A{4:0}` to `/SP`'s `ADR{4:0}`
- `D` to `I`
- `O` to `O`
- `WCLK` to `CLK`
- `WE` to `WE`

### RAM32X1D -> 2x RAMD32

Cell Rules:
- RAMD32 (D6LUT/B6LUT) with `/SP` appended to name
- RAMD32 (C6LUT/A6LUT) with `/DP` appended to name
- No MUXF cells
- 2 outputs

Parameter Rules:
- `IS_WCLK_INVERTED` to `IS_CLK_INVERTED`

Port Rules:
- `A{4:0}` to `/SP`'s `RADR{4:0}`
- `A{4:0}` to `/SP`'s `WADR{4:0}`
- `DPRA{4:0}` to `/DP`'s `RADR{4:0}`
- `A{4:0}` to `/DP`'s `WADR{4:0}`
- `D` to all `I`
- `SPO` to `/SP`'s `O`
- `DPO` to `/DP`'s `O`
- `WCLK` to all `CLK`
- `WE` to all `WE`

### RAM32M -> 2x RAMS32 + 6x RAMD32

Cell Rules:
- RAMS32 (D5LUT) with `/RAMD` appended to name
- RAMS32 (D6LUT) with `/RAMD_D1` appended to name
- RAMD32 (C5LUT) with `/RAMC` appended to name
- RAMD32 (C6LUT) with `/RAMC_D1` appended to name
- RAMD32 (B5LUT) with `/RAMB` appended to name
- RAMD32 (B6LUT) with `/RAMB_D1` appended to name
- RAMD32 (A5LUT) with `/RAMA` appended to name
- RAMD32 (A6LUT) with `/RAMA_D1` appended to name
- No MUXF cells
- 8 outputs

Parameter Rules:
- `IS_WCLK_INVERTED` to `IS_CLK_INVERTED`

Port Rules:
- `WCLK` to all `CLK`
- `WE` to all `WE`
- `DIA[0]` to `/RAMA`'s `I`
- `DIA[1]` to `/RAMA_D1`'s `I`
- `DIB[0]` to `/RAMB`'s `I`
[... 138 lines cut ...]
- RAMS64E (B6LUT) with `/RAMS64E_B` appended to name
- RAMS64E (A6LUT) with `/RAMS64E_A` appended to name
- MUXF7 with `/F7.A` appended to name
- MUXF7 with `/F7.B` appended to name
- MUXF8 with `/F8` appended to name for single output

Parameter Rules:
- `IS_WCLK_INVERTED` to `IS_CLK_INVERTED`

Port Rules:
- `WCLK` to all `CLK`
- `WE` to all `WE`
- `D` to all `I`
- `A[5:0]` to all `ADR{5:0}`
- `A[7:6]` to all `WADR{7:6}`
- `A[6]` to `/F7.A`'s `S`
- `A[6]` to `/F7.B`'s `S`
- `A[7]` to `/F8`'s `S`
- `/RAMS64E_D`'s `O` to `/F7.B`'s `I0`
- `/RAMS64E_C`'s `O` to `/F7.B`'s `I1`
- `/RAMS64E_B`'s `O` to `/F7.A`'s `I0`
- `/RAMS64E_A`'s `O` to `/F7.A`'s `I1`
- `/F7.B`'s `O` to `/F8`'s `I0`
- `/F7.A`'s `O` to `/F8`'s `I1`
- `O` to `/F8`'s `O`

hansemro, 2024-02-08:
## LUTRAM Primitive to BEL Transformations:

### RAMD64E -> LUT_OR_MEM6 BEL

- nextpnr-specific rules:
    - set type to `id_SLICE_LUTX`
    - transform parameter `IS_CLK_INVERTED` to `IS_WCLK_INVERTED`
    - set attribute `X_LUT_AS_DRAM` to `1`
- `RADR{5:0}` to `A{6:1}`
- `WADR{7:0}` to `WA{8:1}`
- `I` to `DI1`
- `O` to `O6`
- `WE` to `WE`
- `CLK` to `CLK`

### RAMS64E -> LUT_OR_MEM6 BEL

- nextpnr-specific rules:
    - set type to `id_SLICE_LUTX`
    - transform parameter `IS_CLK_INVERTED` to `IS_WCLK_INVERTED`
    - set attribute `X_LUT_AS_DRAM` to `1`
- `ADR{5:0}` to `A{6:1}`
- `ADR{5:0}` to `WA{6:1}`
- `WADR{7:6}` to `WA{8:7}`
- `I` to `DI1`
- `O` to `O6`
- `WE` to `WE`
- `CLK` to `CLK`

### RAMD32 -> LUT_OR_MEM6 BEL

- nextpnr-specific rules:
    - set type to `id_SLICE_LUTX`
    - transform parameter `IS_CLK_INVERTED` to `IS_WCLK_INVERTED`
    - set attribute `X_LUT_AS_DRAM` to `1`
- `RADR{4:0}` to `A{5:1}`
- `WADR{4:0}` to `WA{5:1}`
- `I` to `DI2`
- `O` to `O6`
- `WE` to `WE`
- `CLK` to `CLK`

### RAMD32 -> LUT_OR_MEM5 BEL

- nextpnr-specific rules:
    - set type to `id_SLICE_LUTX`
    - transform parameter `IS_CLK_INVERTED` to `IS_WCLK_INVERTED`
    - set attribute `X_LUT_AS_DRAM` to `1`
- `RADR{4:0}` to `A{5:1}`
- `WADR{4:0}` to `WA{5:1}`
- `I` to `DI1`
- `O` to `O5`
- `WE` to `WE`
- `CLK` to `CLK`

### RAMS32 -> LUT_OR_MEM6 BEL

- nextpnr-specific rules:
    - set type to `id_SLICE_LUTX`
    - transform parameter `IS_CLK_INVERTED` to `IS_WCLK_INVERTED`
    - set attribute `X_LUT_AS_DRAM` to `1`
- `ADR{4:0}` to `A{5:1}`
- `ADR{4:0}` to `WA{5:1}`
- `I` to `DI2`
- `O` to `O6`
- `WE` to `WE`
- `CLK` to `CLK`

### RAMS32 -> LUT_OR_MEM5 BEL

- nextpnr-specific rules:
    - set type to `id_SLICE_LUTX`
    - transform parameter `IS_CLK_INVERTED` to `IS_WCLK_INVERTED`
    - set attribute `X_LUT_AS_DRAM` to `1`
- `ADR{4:0}` to `A{5:1}`
- `ADR{4:0}` to `WA{5:1}`
- `I` to `DI1`
- `O` to `O5`
- `WE` to `WE`
- `CLK` to `CLK`

hansfbaier, 2024-02-08:
Thanks for the effort! I am looking forward to what you will come up with!

hansemro, 2024-02-12:
## Issue: nextpnr checks INIT{A:D} instead of INIT_{A:D} parameters for RAM32M/RAM64M

While working on an INIT parameter test, I noticed that the INIT parameters for RAM32M/RAM64M were not being set in the FASM result.

https://github.com/openXC7/nextpnr-xilinx/blob/1c57f511f80945a709d1d43478d39ad0b6cd51d2/xilinx/pack_dram.cc#L455-L469

Merely adding underscores does not immediately solve the issue, so I will need to look more into this later.

hansemro, 2024-02-13:
> Issue: nextpnr checks INIT{A:D} instead of INIT_{A:D} parameters for RAM32M/RAM64M

This should be fixed in this branch: https://github.com/hansemro/nextpnr-xilinx/tree/fix-ram32m-ram64m-init

However, I am noticing some discrepancies compared to Vivado that I will need to verify. Notice how the upper and lower 32-bits of the ?LUT.INIT pattern are swapped.

NextPNR fasm result:
```
CLBLM_L_X84Y126.SLICEM_X0.ALUT.INIT[63:0] = 64'b0000000000000000111011100100010000000000000000000101000001010000
CLBLM_L_X84Y126.SLICEM_X0.ALUT.DI1MUX.AI
CLBLM_L_X84Y126.SLICEM_X0.ALUT.SMALL
CLBLM_L_X84Y126.SLICEM_X0.ALUT.RAM
CLBLM_L_X84Y126.SLICEM_X0.AOUTMUX.O5
CLBLM_L_X84Y126.SLICEM_X0.BLUT.INIT[63:0] = 64'b0000000000000000111011100100010000000000000000001111101011111010
CLBLM_L_X84Y126.SLICEM_X0.BLUT.DI1MUX.BI
CLBLM_L_X84Y126.SLICEM_X0.BLUT.SMALL
CLBLM_L_X84Y126.SLICEM_X0.BLUT.RAM
CLBLM_L_X84Y126.SLICEM_X0.BOUTMUX.O5
CLBLM_L_X84Y126.SLICEM_X0.CLUT.INIT[63:0] = 64'b0000000000000000000100011011101100000000000000001010111110101111
CLBLM_L_X84Y126.SLICEM_X0.CLUT.DI1MUX.CI
CLBLM_L_X84Y126.SLICEM_X0.CLUT.SMALL
CLBLM_L_X84Y126.SLICEM_X0.CLUT.RAM
CLBLM_L_X84Y126.SLICEM_X0.COUTMUX.O5
CLBLM_L_X84Y126.SLICEM_X0.DLUT.INIT[63:0] = 64'b0000000000000000000100011011101100000000000000000000010100000101
CLBLM_L_X84Y126.SLICEM_X0.DLUT.SMALL
CLBLM_L_X84Y126.SLICEM_X0.DLUT.RAM
CLBLM_L_X84Y126.SLICEM_X0.DOUTMUX.O5
CLBLM_L_X84Y126.SLICEM_X0.WEMUX.CE
CLBLM_L_X84Y126.SLICEM_X0.NOCLKINV
CLBLM_L_X84Y126.SLICEL_X1.NOCLKINV
```

Vivado bit2fasm result:
```
CLBLM_L_X48Y95.SLICEL_X1.NOCLKINV
CLBLM_L_X48Y95.SLICEL_X1.PRECYINIT.C0
CLBLM_L_X48Y95.SLICEM_X0.ALUT.DI1MUX.BDI1_BMC31
CLBLM_L_X48Y95.SLICEM_X0.ALUT.INIT[46:0] = 47'b10100000101000000000000000000001110111001000100
CLBLM_L_X48Y95.SLICEM_X0.ALUT.RAM
CLBLM_L_X48Y95.SLICEM_X0.ALUT.SMALL
CLBLM_L_X48Y95.SLICEM_X0.AOUTMUX.O5
CLBLM_L_X48Y95.SLICEM_X0.BLUT.DI1MUX.DI_CMC31
CLBLM_L_X48Y95.SLICEM_X0.BLUT.INIT[47:0] = 48'b111110101111101000000000000000001110111001000100
CLBLM_L_X48Y95.SLICEM_X0.BLUT.RAM
CLBLM_L_X48Y95.SLICEM_X0.BLUT.SMALL
CLBLM_L_X48Y95.SLICEM_X0.BOUTMUX.O5
CLBLM_L_X48Y95.SLICEM_X0.CLUT.DI1MUX.DI_DMC31
CLBLM_L_X48Y95.SLICEM_X0.CLUT.INIT[47:0] = 48'b101011111010111100000000000000000001000110111011
CLBLM_L_X48Y95.SLICEM_X0.CLUT.RAM
CLBLM_L_X48Y95.SLICEM_X0.CLUT.SMALL
CLBLM_L_X48Y95.SLICEM_X0.COUTMUX.O5
CLBLM_L_X48Y95.SLICEM_X0.DLUT.INIT[42:0] = 43'b1010000010100000000000000000001000110111011
CLBLM_L_X48Y95.SLICEM_X0.DLUT.RAM
CLBLM_L_X48Y95.SLICEM_X0.DLUT.SMALL
CLBLM_L_X48Y95.SLICEM_X0.DOUTMUX.O5
CLBLM_L_X48Y95.SLICEM_X0.NOCLKINV
CLBLM_L_X48Y95.SLICEM_X0.PRECYINIT.C0
CLBLM_L_X48Y95.SLICEM_X0.WEMUX.CE
```

hansemro, 2024-02-13:
## Issue: nextpnr does nothing with `IS_*CLK_INVERTED` property for LUTRAM cells

While working on CLKINV property test, I noticed that nextpnr does not set the CLKINV bit when `IS_*CLK_INVERTED` property for a LUTRAM is set. Instead, nextpnr fasm writer sets NOCLKINV for the SLICEM site.

WIP CLKINV property test: https://github.com/hansemro/primitive-tests/tree/xc7-lutram-tests/lutram-tests/clkinv-test

hansfbaier, 2024-02-17:
Great work!

hansemro, 2024-08-30:
Something that bothered me about how RAM{S,D}64E maps to LUT_OR_MEM6 BEL is that only DI1 data input is used to write to both internal LUT_OR_MEM5 BELs. Somehow one of the LUT_OR_MEM5 BEL can select between DI1 and DI2 data inputs, but this is not really well documented.

Recently, I stumbled on the following physical design rules (in `$VIVADO_2017.2_ROOT/ids_lite/ISE/msg/usenglish/PhysDesignRules.msg`) that seem to correspond to RAM.SMALL configuration bit being what controls data input selection:

```
1383
Issue with pin connections and/or configuration on block:<%s>:<%s>.  For RAMMODE programming set with DPRAM64 or SPRAM64 or SRL32 the DI1 input pin must be connected.\n
Issue with pin connections and/or configuration on block:<!%1!>:<!%2!>.  For RAMMODE programming set with DPRAM64 or SPRAM64 or SRL32 the DI1 input pin must be connected.\n
;;
1384
Issue with pin connections and/or configuration on block:<%s>:<%s>.  For RAMMODE programming set with DPRAM64 or SPRAM64 or SRL32 the DI2 input pin cannot be connected.\n
Issue with pin connections and/or configuration on block:<!%1!>:<!%2!>.  For RAMMODE programming set with DPRAM64 or SPRAM64 or SRL32 the DI2 input pin cannot be connected.\n
;;
1385
Issue with pin connections and/or configuration on block:<%s>:<%s>.  For RAMMODE programming set with DPRAM32 or SPRAM32 or SRL16 the DI2 input pin must be connected.\n
Issue with pin connections and/or configuration on block:<!%1!>:<!%2!>.  For RAMMODE programming set with DPRAM32 or SPRAM32 or SRL16 the DI2 input pin must be connected.\n
;;
1386
Issue with pin connections and/or configuration on block:<%s>:<%s>.  For RAMMODE programming set with DPRAM32 or SPRAM32 or SRL16 the DI1 input pin cannot be connected.\n
Issue with pin connections and/or configuration on block:<!%1!>:<!%2!>.  For RAMMODE programming set with DPRAM32 or SPRAM32 or SRL16 the DI1 input pin cannot be connected.\n
```

Coincidentally, [018-clb-ram prjxray fuzzer](https://github.com/f4pga/prjxray/blob/master/fuzzers/018-clb-ram/README.md#clb-ram-fuzzer) found that RAM.SMALL configuration bit is set for RAM32M/RAM32X1{S,D} and SRL16E, but not set for RAM64M/RAM{64,128}X{S,D}/RAM256X1S and SRLC32E. This all seems to indicate RAM.SMALL bit is used for data input selection for upper LUT_OR_MEM5 BEL (one with DI2 input and initialized with INIT[63:32]).

Here is a block diagram to help visualize what I see of LUT_OR_MEM6 BEL:

[image]

hansfbaier, 2024-08-31:
Good to see you making progress!

lehaifeng000, 2024-09-01:
@hansemro the DI2 can't be used, it would make the Infinite loop in place. there is some check in the place, if the lutram used the DI2 port, and the wa7-8 ports are not being config, It cannot complete the place

hansemro, 2024-09-02:
@lehaifeng000 Yes, nextpnr does not currently have the capacity to pack/place RAM32X1S/RAMS32 which would occupy and utilize both LUT_OR_MEM5 BELs. As you say, placer will get stuck because LUT_OR_MEM5/RAM32* with DI2 is not yet accepted as legal: https://github.com/openXC7/nextpnr-xilinx/blob/9debb871624163d4043150e576793c78cba503f8/xilinx/arch_place.cc#L114-L117.

Additionally, how {C,B}X pins are connected to {C,B}LUTs and WA7USED/WA8USED BEL should also be considered in legalization. I believe, Vivado avoids this by making it illegal to place mismatched LUTRAM types in the same CLB SLICEM site. However, I will need to look more into this.

If we update the packer to utilize DI2, we will need to and should update the placer and legalization accordingly.

lehaifeng000, 2024-09-02:
I tend to understand and try to modify place, inserting rules into the placement process.

lehaifeng000, 2024-09-02:
@hansfbaier @hansemro 
I'm wondering if you could provide an email or other contact information. Sometimes, materials that aren't suitable for public sharing can be sent to you privately.

hansfbaier, 2024-09-02:
@lehaifeng000 my email address foss@hans-baier.de should be visible in every git commit. Same for @hansemro

lehaifeng000, 2024-09-02:
> @lehaifeng000 my email address [foss@hans-baier.de](mailto:foss@hans-baier.de) should be visible in every git commit. Same for @hansemro
OK, I got it

hansfbaier, 2024-10-13:
@hansemro Are the changes in your branch conflicting with the primitive-tests branch?
Can you make PRs for the elements you implemented that already pass tests?

cheungxi, 2025-08-07:
Has this issue been fixed?

hansemro, 2025-08-08:
@cheungxi No. This issue is present and currently out of my depth to handle appropriately. Reading more into packing/placing/routing made me realize there is much more to consider than to simply adding support for larger LUTRAMs such as support for packing/placing multiple LUT(RAM)s together in the same SLICE site while considering congestion/routability [1][2], or allowing unpacking/repacking of cells to explore more of the placement/routing solution space[3]. Maybe this could be treated as an extension of basic support where we initially place each LUT(RAM) in their own SLICE(M) site then introduce multi-LUT(RAM) packing and even fixed SITE+BEL placement with proper legalization rules. I am also fine with this approach to narrow the scope and make more smaller and progressive changes.

@hansfbaier @lehaifeng000 I apologize for not having communicated these concerns clearly and early, and may have stalled community progress on this issue. It was very hard for me know exactly what to aim for then and decide what to spend time on (DRC-based validation vs build synthetic test cases, experiment placement algorithms with RapidWright, etc.). Anyhow, I think it may be best for someone else (better if already familiar with this domain) to take over.

[1] ISPD 2016 Routability-Driven FPGA Placement Contest: https://www.ispd.cc/contests/16/ispd2016_contest.html
[2] RippleFPGA: https://ieeexplore.ieee.org/document/7827644
[3] GPlace3.0 presentation+slides: https://www.youtube.com/watch?v=bd57F-yG31Y
    - Unfortunately, audio quality is quite poor but offers a good introduction to the challenges/solutions involved; I used easy effects with some filters + reverb to make it somewhat tolerable to listen

hansfbaier, 2026-09-19:
## Status of this issue, 2026-09-19 (verified locally against `main` = `6811cb8e`)

I went through the whole list in the body against current `main`, with a survey of 34 designs built
with the toolchain's yosys (0.67) and a nextpnr-xilinx built from `main` itself. Everything below is
reproduced, none of it is inferred from the code alone; the raw table and the case generator are in
the comment at the end.

**Short version:** the *inference* half of this issue is done — every distributed-RAM macro yosys can
emit for xc7 now packs, places and produces a FASM. What is left are the primitives only a
hand-written or Vivado-imported netlist can contain, plus one silent wrong-bitstream bug found while
testing them (`RAM64X1S_1` / `IS_WCLK_INVERTED`), now split out as #201 and #202.

### The nine primitives in this issue's description

| Primitive | State on `main` today | Notes |
|---|---|---|
| `RAMS32` (manual) | **works** | handled by the raw-leaf arm of `pack_dram.cc` (`ADR` mirrored onto `WADR`, then the `RAMD32` rules) |
| `RAMD32` (manual) | **works** | same arm |
| `RAMS64E` (manual) | **unsupported** | `ERROR: Unable to place cell 'ram', no Bels remaining of type 'RAMS64E'` — no arm anywhere |
| `RAMD64E` (manual) | **broken** | packs, then `ERROR: No wire found for port CLK on destination cell ram` in post-placement legalisation; `RAMD32` through the same arm is fine |
| `RAM32X1S` | **unsupported** | `ERROR: Cannot pack unsupported primitive: RAM32X1S` |
| `RAM64X1S` | **fixed 2026-09-19** (#196) | inferred for *any* 64-deep memory, `w1` and `w8` verified; 8 LUT-RAMs in 2 slices (4+4), no `WA7USED`/`WA8USED` |
| `RAM64X1S_1` (inverted clock) | **broken, and worse than "unsupported"** | as a cell: `no Bels remaining of type 'RAM64X1S_1'`; as a parameter (`IS_WCLK_INVERTED=1`): **silently dropped**, FASM byte-identical to the non-inverted design and `NOCLKINV` programmed — wrong clock edge with exit 0. Tracked in #201 |
| `RAM128X1S` | **fixed 2026-08-13** (#144) | 16 LUT-RAMs in 8 slices (2+2) + `WA7USED`, `w1`/`w8`, scalar `A0..A6` |
| `RAM256X1S` | **fixed 2026-08-13** (#142) | 32 LUT-RAMs in 8 slices (4+4) + `WA7USED`/`WA8USED`, `w1`/`w8` |

### What the survey covers beyond the list

* Inference, all **ok**: 16/32-deep single port (→ `RAM32M`), 64-deep (→ `RAM64X1S`), 128-deep
  (→ `RAM128X1S`), 256-deep (→ `RAM256X1S`), `w1` and `w8` each; 64-deep with a **negedge** clock
  (yosys inserts a fabric inverter and emits a plain `RAM64X1S` — no slice inversion needed);
  simple-dual-port 64/128/256 (→ `RAM64M`/`RAM128X1D`/4×`RAM64M`); quad-port 64 (→ `RAM64M`).
* A true dual-port memory (two write ports, two clocks) is rejected by **yosys** (`no valid mapping
  found`), i.e. before nextpnr — not a gap here.
* Manual instantiation, all **ok**: `RAM32X1D`, `RAM64X1D`, `RAM128X1D`, `RAM32M`, `RAM64M`.
  Correctly rejected (Ultrascale-only): `RAM256X1D`, `RAM512X1S`.
* Multi-LUT packing (raised in the hand-off below): already works — 4 LUT-RAMs per SLICEM for
  64-deep and 256-deep shapes, 2 per slice for 128-deep (data + write-address decoder), and 4
  sub-DRAMs per site for `RAM32M`/`RAM64M`.
* Reachability: `share/yosys/xilinx/lutrams_xc5v_map.v` instantiates only `RAM32M`, `RAM32M16`,
  `RAM64M`, `RAM64M8`, `RAM64X1S`, `RAM64X1D`, `RAM128X1S`, `RAM128X1D`, `RAM256X1S`, `RAM256X1D`,
  `RAM512X1S` — no `RAM32X1S`, no `*_1`. So the remaining names in this issue are import/hand-
  instantiation gaps; no synthesisable xc7 design hits them today.

### Release state — the 64-deep fix is in no release yet

`0.9.5` (`a9ceeec2`) contains #142/#144 (128/256-deep work, verified: those cases pass on the
released binary) but **not** #196. On 0.9.5, `reg [W-1:0] m [0:63]` — the most common LUTRAM shape —
still aborts with `Cannot pack unsupported primitive: RAM64X1S`; the released binary also fails the
repo's own `demo-projects/regression/lutram-ram64x1s` case, which `main` passes. Same for every
0.9.4-labelled build on hand. A release carrying #196/#197/#199/#200 is being prepared.

### The linked regression

`openXC7/demo-projects#6` (litex-ddr-kc705 + litex-ddr-hpcstore-k420t) is fixed in practice: both
demos are green in CI — `litex-ddr-kc705` in smoke run
[34782068034](https://github.com/openXC7/demo-projects/actions/runs/34782068034) (its build log maps
65 memories via `$__XILINX_LUTRAM_SP_`) and `litex-ddr-hpcstore-k420t` in the 2026-09-18 nightly
heavy run.

### Tasks/Status (replacing the `TODO: rewrite tasks` stub in the body)

- [x] `RAM128X1S`, `RAM256X1S` — #142 (2026-08-13), #144 (2026-08-13); shipped in 0.9.5
- [x] `RAM64X1S` — #196 (2026-09-19); **pending a release**
- [x] `RAMS32`, `RAMD32` (manual) — raw-leaf arm
- [x] Inference coverage: single/double/quad-port, `w1`/`w8`, depths 16/32/64/128/256, negedge
- [ ] `IS_WCLK_INVERTED` / `*_1` variants: currently a **silent wrong bitstream** — #201
- [ ] `RAM32X1S`, `RAMS64E`, `RAMD64E` leaves — #202
- [ ] Re-enable `demo-projects/regression/lutram-ram64x1s` and `const-holdout` (both now pass on
      `main`; they are still switched off)

Given the above, this issue can now be narrowed to the "manual/imported primitives" scope of #201 and
#202, or closed with those two as the successors — your call.

### Method / evidence

Instrument: 34 cases (`inf-*` inference, `man-*` manual instantiation, `ab-*` pin-safe A/B shapes),
each `top.v` + `top.xdc` + `part.txt`, driven by a copy of `demo-projects/regression/run.sh`; yosys
[... cut ...]
128/256-deep cases. Reproduced for every row of the table above; the failures quote nextpnr's own
error text.

One thing I could **not** verify: a runtime before/after for #142/#144. The tree before the
128/256-deep packing is `d5dc0cac` (`9ec2b030^`), and that build rejects every chipdb on hand with
`The internal IDs of nextpnr are inconsistent with the supplied chip database`, so the pre-fix
behaviour is established from source instead — `d5dc0cac:xilinx/pack_dram.cc` lists `RAM64X1S`,
`RAM128X1S` and `RAM256X1S` in the `Cannot pack unsupported primitive` arm.

hansfbaier, 2026-09-19:
Small correction to the survey above, from a verification pass on the current tree.

The line *"0.9.5 (`a9ceeec2`) contains #142/#144 but **not** #196 … A release carrying #196/#197/#199/#200 is being prepared"* is now stale: that release is **0.9.6** (published 2026-09-19 04:13Z, `3fd78784`), and it does carry #196. Checked with the repo's own regression case on `xc7a200tfbg484-2`:

| build | `regression/lutram-ram64x1s` |
|---|---|
| 0.9.5 | exit 255 — `ERROR: Cannot pack unsupported primitive: RAM64X1S` |
| 0.9.6 | exit 0, FASM written (`9ef9f030`) |
| main `d486d8ef` | exit 0, FASM written (`45036216`) |

I also re-ran four rows of the survey independently on `main` and they match: RAM64X1S inferred (8 LUT-RAMs in 2 sites, no WA7/WA8USED), RAM128X1S (16 LUT-RAMs, 8 WA7USED), RAM256X1S (32 LUT-RAMs, 8 WA7USED + 8 WA8USED), and manual RAM32X1S still `Cannot pack unsupported primitive` (tracked in #202).

One row has changed since the survey's base revision `6811cb8e`: the `RAM64X1S_1` / `IS_WCLK_INVERTED=1` entry ("silently dropped, FASM byte-identical, NOCLKINV programmed") no longer reproduces after PR #203 — an inverted `RAM32X1D`/`RAM64X1D`/`RAM64X1S` now programs the half-slice `CLKINV` instead of `NOCLKINV` (see #33: the two designs differ in exactly that one FASM line).