openXC7/nextpnr-xilinx #22: nice to meet you

lehaifeng000, 2024-05-16:
I have implemented most primitives and prjxray info of the XC7 devices using the f4pga-arch-defs project. but I've encountered some challenging issues with VPR, and Xilinx's architecture doesn't seem to fit well with VPR either. I am planning to try nextpnr next. I have forked your project. would it be possible to reach out to you if I encounter any issues in the future?

hansfbaier, 2024-05-16:
Yes, of course, I suggest we work closely together to avoid duplicated effort.
What are your goals?
What are you going to work on next?

hansfbaier, 2024-05-16:
I am currently working on GTP Transceiver support.

lehaifeng000, 2024-05-16:
thank you, I want to start by adding FIFO18E1, and in the process, familiarize myself with the nextpnr codebase.

hansfbaier, 2024-06-19:
I think this issue is done than.
Let me know when you have issues.

closed 2024-06-19 by hansfbaier

lehaifeng000, 2024-06-19:
no, thanks!