openXC7/nextpnr-xilinx #25: BSCANE2 placement not determined by JTAG_CHAIN parameter

hansemro, 2024-06-19:
## Issue Description

NextPNR's placer may place BSCANE2 BELs at a different location than expected, and thus require a different USERx instruction to access.

## Steps to Reproduce

1. Clone `jtag_led` demo:

```
git clone https://github.com/hansemro/jtag_led.git
cd
```

2. Modify USER_PORT/JTAG_CHAIN parameter in `src/rtl/jtag_led.sv` with any value between 1-4 (inclusive).

3. Build with `make` using openXC7 toolchain:

```
nix develop github:openXC7/toolchain-nix
make
```

4. Check if `NEXTPNR_BEL` attribute for `bscane` cell is valid or not inside `jtag_led.place.json`.

### Expected placement:

| `JTAG_CHAIN` | `NEXTPNR_BEL` |
|-----------------|------------------|
| 1 | `BSCAN_X0Y0` |
| 2 | `BSCAN_X0Y1` |
| 3 | `BSCAN_X0Y2` |
| 4 | `BSCAN_X0Y3` |

## Workaround

As a workaround, you can manually correct placement in `jtag_led.place.json` and let nextpnr route with the corrected json:

```
# clear build
make clean

# build 
make jtag_led.place.json

# edit jtag_led.place.json if necessary
...

# let make build the bitstream with corrected placement
make
```

hansfbaier, 2024-06-19:
@hansemro Hi, thanks for this perfect report. That made understanding, reproducing and fixing the problem really easy.
I implemented a fix here:
https://github.com/openXC7/nextpnr-xilinx/commit/50685eb68e877c07fa3fc14722deb06e66725ac3
it is in the pack-bscan branch.
Please test it on your hardware, and let me know if it works,
then I will merge it.

hansemro, 2024-06-19:
> Please test it on your hardware, and let me know if it works,
> then I will merge it.

Currently away, but will test soon. Thanks!

hansemro, 2024-06-19:
@hansfbaier pack-bscan branch resolves this issue. Thanks!

hansfbaier, 2024-06-20:
OK this is merged. Thanks for the great report!

closed 2024-06-20 by hansfbaier