openXC7/nextpnr-xilinx #33: LUTRAM write occurs on rising edge regardless of IS_WCLK_INVERTED property

hansemro, 2024-06-29:
## Issue Description

Issue discovered in https://github.com/openXC7/nextpnr-xilinx/issues/20#issuecomment-1942858331

When IS_WCLK_INVERTED property is asserted, we expect the write to LUTRAM to occur on the falling edge. However, this does not happen as the placer ignores the property for LUTRAMs, and, therefore, treats as if all writes should occur on rising edge. This may lead to functional errors.

## Steps to Reproduce and Test

To distinguish between a write occuring on a positive edge and negative edge of clock, I created a simple test with two LUTRAMs sensitive to different edges of clock. Output of posedge LUTRAM and negedge LUTRAM drives led_o[0] and led_o[1], respectively. A single BSCANE2 primitive will be used to manually drive WCLK+DI+WE pins of the LUTRAM via JTAG.

1. Clone primitive-tests repo fork with `xc7-lutram-negedge-write-demo` branch

```
git clone https://github.com/hansemro/primitive-tests.git -b xc7-lutram-negedge-write-demo
```

2. Navigate to primitive-tests/lutram-tests/negedge-write-test and build the test with the openXC7 toolchain:

> [!NOTE]
> If targeting an FPGA board other than SQRL Acorn CLE215+, then provide an XDC file with
> 3 active-low LEDs mapped to led_o[2:0] and update `BOARD`/`FAMILY`/`PART`/`JTAG_CABLE`/`XDC`
> target parameters in the Makefile.

```
$ cd primitive-tests/lutram-tests/negedge-write-test
$ nix develop github:openXC7/toolchain-nix
[nix(openXC7)] make
```

3. Check that `CLKINV` bit is not set anywhere inside `top.fasm` (a sign that the test will fail). Proceed with the following steps to test.

4. Load the bitstream onto the FPGA:

```
make program
```

5. With LUTRAM initialized to 0, write a 1 on positive edge with the following openocd command:

> [!NOTE]
> Use an appropriate interface adapter script (usually located in `/usr/share/openocd/scripts/interface/`).
>
> Digilent HS2 support is provided with `digilent-hs2.cfg` in case `interface/ftdi/digilent-hs2.cfg` does not work.

> [!NOTE]
> `setup.cfg` defines a function `set_lutram <we> <data> <clk>` to manually set LUTRAM input pins.
>
> To avoid setup/hold violations, we should not change clk and we/data together.

```
openocd -f interface/ADAPTER.cfg -f setup.cfg \
    -c "set_lutram 0 0 0" \
    -c "set_lutram 0 0 0" \
    -c "set_lutram 1 1 0" \
    -c "set_lutram 1 1 1" \
    -c "shutdown"
```

If the design works correctly, only the LED (led_o[0]) assigned to the posedge LUTRAM should be lit after a positive-edge write. If LEDs led_o[1:0] light up together, then the design was incorrectly implemented. (LED at led_o[2] should dim on positive-edge write since the outputs should differ).

Anyhow, to perform a write on the negative edge, we would run the following afterwards:

```
openocd -f interface/ADAPTER.cfg -f setup.cfg \
    -c "set_lutram 1 1 0"
    -c "shutdown"
```

6. (Optional) Compare positive-edge (and negative-edge) write behavior against bitstream built with Vivado bitstream:

```
make vivadoclean top.vivado.bit
make BITSTREAM=top.vivado.bit program
# initial positive-edge write
openocd -f interface/ADAPTER.cfg -f setup.cfg \
    -c "set_lutram 0 0 0" \
    -c "set_lutram 0 0 0" \
    -c "set_lutram 1 1 0" \
    -c "set_lutram 1 1 1" \
    -c "shutdown"
# negative-edge write
openocd -f interface/ADAPTER.cfg -f setup.cfg \
    -c "set_lutram 1 1 0"
    -c "shutdown"
```

Design implemented with Vivado should have the two LUTRAM perform writes at different edges of the clock.

## Other Considerations

LUTRAM and FFs share clock from CLKINV routing BEL, so all synchronous elements within the SLICE site must have the same IS_*CLK_INVERTED property.

hansfbaier, 2024-06-29:
Do you know how to disassemble a bitstream with bit2fasm?

hansemro, 2024-06-29:
> Do you know how to disassemble a bitstream with bit2fasm?

Yes, here:

```
bit2fasm --db-root ${PRJXRAY_DB_DIR}/${FAMILY}/ --part ${PART} top.bit > top.fasm
```

hansfbaier, 2024-06-30:
Not sure if I am comparing the right two parts here, but the CLKINV/NOCLKINF bits probably make the difference
[image]

hansfbaier, 2024-06-30:
See `write_luts_config` and `write_ffs_config` in `fasm.cc`

hansfbaier, 2024-06-30:
It might be a case for LUTRAMs is missing here:
[image]

hansemro, 2024-06-30:
@hansfbaier Yes, LUTRAM is currently not being handled in the fasm writer. Since LUTRAMs and FFs share the CLKINV BEL, I am thinking that the CLKINV bit should not be set inside `write_ffs_config` but inside a new function that handles both LUTRAMs and FFs. This new function would also be able to do some final DRC to check that the clock inversion property value matches.

hansemro, 2024-06-30:
There is a rule in place (https://github.com/openXC7/nextpnr-xilinx/blob/30ba8fbe77ace38e2e4ff08e9251d1e0d203c11d/xilinx/arch_place.cc#L557-L558) to check that the FFs have the same inversion property. We may want to define a new rule to compare LUTRAM's clkinv and FF's clkinv.

lehaifeng000, 2024-08-30:
yes, I would start to do this next week

hansemro, 2024-08-30:
@lehaifeng000 @hansfbaier

Sounds good. I am still in the process of figuring out how to fully test whether the CLB placement rules work to catch problematic physical implementations. I suppose this is a good time to ask if anyone can reliably place LUTRAMs and FFs in the same CLB slice since I am failing to do this task with openxc7 flow.

hansfbaier, 2024-08-31:
@hansemro How do you try to place them? xdc file, LOC annotations or other means?

lehaifeng000, 2024-09-02:
In the place phase, is there any rule to avoid ff with different clkinv inside a slice?

hansemro, 2024-09-02:
> @hansemro How do you try to place them? xdc file, LOC annotations or other means?

This may be an impossible task for the moment since we do not yet support manual BEL placement yet with LOC and BEL attributes. I was looking into setting CONSTR_{X,Y,Z} attributes in the post-pack json, but I am not entirely sure this is where we should be looking. Sorry for giving a rather loaded task, I don't know enough about nextpnr to give proper instructions.

lehaifeng000, 2024-09-03:
> @hansemro How do you try to place them? xdc file, LOC annotations or other means?

After reviewing the code for "place", I believe a potential optimization could be implemented by modifying the "overused" condition within the "CutSpreader" function.

hansfbaier, 2024-10-13:
@hansemro The primitives-fixes branch contains changes which seem to fix this issue.
Can you confirm that? What do your tests say?

hansfbaier, 2026-09-19:
Verified fixed at the bitstream level — closing, with one silicon caveat at the end.

The defect was that a LUT-RAM's `IS_WCLK_INVERTED` never reached the half-slice clock inverter. PR #203 (`e35b57af`, *drive the half-slice CLKINV from the LUT-RAM clock inversion*, in `main` and in the 0.9.6 release) folds the LUT-RAM cells into the same `is_clkinv` decision as the flip-flops: `xilinx/fasm.cc` now tracks `mem_clkinv` and flags a memory/FF disagreement (`mem_and_ff_disagree`).

Measured on `xc7a35tcsg324-1` with two otherwise identical `RAM32X1D` designs, one with `IS_WCLK_INVERTED=1`:

    diff ram32x1d_plain/top.fasm ram32x1d_inv/top.fasm
    13c13
    < CLBLM_R_X41Y57.SLICEM_X0.NOCLKINV
    ---
    > CLBLM_R_X41Y57.SLICEM_X0.CLKINV

One bit, on the RAM's own half-slice — the parameter now reaches the shared slice inverter instead of being ignored (previously the two FASM files were byte-identical). The negedge case lands correctly too: the two edge-sensitive RAMs of the repro program opposite bits in different slices (`…X41Y64.SLICEM_X0.NOCLKINV` for the posedge RAM, `…X41Y58.SLICEM_X0.CLKINV` for the negedge one). The canonical regression case `demo-projects/regression/lutram-clkinv` (`RAM64X1S` with `IS_WCLK_INVERTED=1`) passes its `check.sh`: `ok: SLICEM_X0 CLKINV with a 64x1 LUT-RAM (no NOCLKINV, no SMALL, no WA7/WA8USED)`.

Silicon caveat: the runtime check in the report (BSCANE2 + JTAG, which LED lights on which edge) needs a board, and the BSCANE2 path still does not build to a bitstream on this tree (#126), so the falling-edge *write* was not confirmed on hardware — only the bitstream bit that selects the edge. If the board test still shows the wrong edge with that bit now set, reopen and I will chase the remaining piece.

closed 2026-09-19 by hansfbaier