openXC7/nextpnr-xilinx #39: nextpnr failed to route 6 RAMB36E1 design, gives ERROR: Failed to route arc 0 of net 'modules_2.mem_ext.Memory.0.8.genblk1.genblk1.CAS_B', from SITEWIRE/RAMB36_X3Y21/CASCADEOUTB to SITEWIRE/RAMB36_X2Y28/CASCADEINB

mirekez, 2024-08-22:
[ERROR_Failed_to_route_arc_0_of_net_modules_2_mem_ext_Memory_0_8_genblk1_genblk1_CAS_B.zip](https://github.com/user-attachments/files/16709343/ERROR_Failed_to_route_arc_0_of_net_modules_2_mem_ext_Memory_0_8_genblk1_genblk1_CAS_B.zip)

Info: Device utilization:                                                                                                                                     
Info:             SLICE_LUTX: 15157/126800    11%                                                                                                             
Info:              SLICE_FFX:   332/126800     0%                                                                                                             
Info:                 CARRY4:  1026/15850     6%                                                                                                              
Info:             PSEUDO_GND:     1/30932     0%                                                                                                              
Info:             PSEUDO_VCC:     1/30932     0%                                                                                                              
Info:                  HARD0:     0/ 2376     0%                                                                                                              
Info:               RAMB18E1:     0/  270     0%                                                                                                              
Info:               FIFO18E1:     0/  135     0%                                                                                                              
Info:           RAMBFIFO36E1:     0/  135     0%                                                                                                              
Info:               RAMB36E1:     6/  135     4%                                                                                                              
Info:                DSP48E1:     5/  240     2%                            

Vivado works (resource usage is different in Vivado, more BRAMS are used)

mirekez, 2024-08-22:
adding very small design with same error, probably will 
[ERROR_Failed_to_route_arc_0_of_net_modules_3_mem_ext_Memory_0_0_genblk1_genblk1_CAS_A.zip](https://github.com/user-attachments/files/16711031/ERROR_Failed_to_route_arc_0_of_net_modules_3_mem_ext_Memory_0_0_genblk1_genblk1_CAS_A.zip)
be easier to debug

Info: Device utilisation:                                                                                                                                     
Info:             SLICE_LUTX:   543/126800     0%                                                                                                             
Info:              SLICE_FFX:   108/126800     0%                                                                                                             
Info:                 CARRY4:    44/15850     0%                                                                                                              
Info:             PSEUDO_GND:     1/30932     0%                                                                                                              
Info:             PSEUDO_VCC:     1/30932     0%                                                                                                              
Info:                  HARD0:     0/ 2376     0%                                                                                                              
Info:               RAMB18E1:     0/  270     0%                                                                                                              
Info:               FIFO18E1:     0/  135     0%                                                                                                              
Info:           RAMBFIFO36E1:     0/  135     0%                                                                                                              
Info:               RAMB36E1:     4/  135     2%

hansfbaier, 2026-09-19:
## Still reproducible on current main — and this one is a real bug, with a fix

Re-ran both attachments (Makefile says `xc7a100tcsg324-2`, Arty A7) on current main / 0.9.6,
with a chipdb from today's prjxray-db and one rebuilt from the rev pinned the day this was
filed (`77938473`) — identical results, so it is not a database artifact:

| toolchain | result |
|---|---|
| 0.9.1 / 0.9.2 | FAIL earlier, on an unrelated net (`$div$…div_mod`) |
| **0.9.3 – 0.9.6, current main** | **FAIL with the reported error** |

* small case: `modules_3.mem_ext.Memory.0.0…CAS_A`, `RAMB36_X0Y29/CASCADEOUTA` → `RAMB36_X0Y20/CASCADEINA` (reported: `X0Y28` → `X0Y24`)
* big case: fails at `modules_2.mem_ext.Memory.0.9…CAS_B`

### Root cause

Each wide memory is mapped to a RAMB36E1 pair with a hard data cascade
(`lower` CASCADEOUT{A,B} → `upper` CASCADEIN{A,B}); the sink has to sit on the one BRAM tile
above the driver. That is a dedicated route and **nothing constrains it**: `pack.cc` only has
the workaround that *disconnects* CASCADEIN pins tied to constants, and the pip blacklist in
`arch.cc` covers only the *address*-cascade plumbing (`CASCOUT`/`CASCINTOP`/`CASCINBOT`) — the
data-cascade pip is in the db and routes fine, which I verified by hand-pinning a pair:

* driver `RAMB36_X0Y25`, sink `RAMB36_X0Y26` → the design **builds**, one hop per cascade
  (`BRAM_L_X6Y125/BRAM_FIFO36_CASCADEOUTA → BRAM_L_X6Y130/BRAM_FIFO36_CASCADEINA`)
* the same pair with the sink one tile *below* the driver → still fails, i.e. direction matters
  (the pip only runs upward)

So the placer scattering the pair (memory 0.0 landed 9 sites apart; memory 0.1 landed
**inverted**) is the whole failure. The DSP P cascade doesn't have this problem because
`pack_dsp_xc7.cc:walk_dsp()` chains it; there is no equivalent for RAMB36E1.

### Fix

A placer chain constraint is not usable here: the relation is **not** a constant offset — BRAM
tile rows step 5 *or* 6 in y (108 hops of `-5`, 22 of `-6` on xc7a100t), so anything expressed
as `constr_y` would be wrong for some rows, and HeAP aborts the build outright when a cluster
cannot be legalised. I put it in the post-placement repair path instead
(`Arch::fixupBramCascades()`, called at the end of `fixupPlacement()` in `arch_place.cc`): for
every RAMB36E1 whose cascade output drives another RAMB36E1's cascade input, walk the chipdb
from the *placed* driver's bel pin to the only `CASCADEIN` bel it reaches and relocate the sink
there (swap when that bel is occupied by another movable RAMB cell; warn and leave alone for
stamped/pinned cells). +148 lines in `xilinx/arch.h` + `xilinx/arch_place.cc`, no changes to
placement or routing behaviour otherwise.

Verified locally:

* both attachments from this issue now build: 0 errors, FASM 566 851 B (small) / 6 872 184 B
  (big), cascade routed as a single dedicated hop each (`BRAM_L_X6Y145/BRAM_FIFO36_CASCADEOUTA
  → BRAM_L_X6Y150/BRAM_FIFO36_CASCADEINA` for one of them); correctly placed pairs are left
  alone (`2 link(s) already legal, 2 relocated`)
* no change to anything else: the four certified arty-s7 demos are byte-identical to 0.9.6
  (litex-ddr `7938cc6f…`, litex-minimal `dbe6e788…`, ddr3-test `10a15226…`, dsp-test
  `289fd897…`), and the DSP-cascade design from #35 is unchanged too (`efde18ee…` both) — the
  new pass finds no cascade links in those designs at all, so it is a no-op there

Not verified on silicon — the design targets an Arty A7-100T and I have no A7 board.

The patch is written and verified locally (+148 lines in `xilinx/arch.h` + `xilinx/arch_place.cc`)
but not pushed — happy to open a PR if a repair pass is the shape you want, or to turn it into a
`walk_dsp`-style packer constraint plus a fixed-up geometry if you'd rather keep it in the packer.

Reproduce either case with:

```sh
pypy3 bbaexport.py --device xc7a100tcsg324-2L --bba a.bba && bbasm -l a.bba xc7a100tcsg324.bin
nextpnr-xilinx --chipdb xc7a100tcsg324.bin --xdc TestPipeline.xdc --json TestPipeline.json --fasm out.fasm
```

closed 2026-09-19 by hansfbaier