openXC7/nextpnr-xilinx #42: warning: format '%d' expects on Arm64 M1Mac

hsk, 2024-08-27:
Replace %d with %ld because a warning will appear and the build will stop.

```
24.66 /nextpnr-xilinx/bba/main.cc: In function 'int main(int, char**)':
24.66 /nextpnr-xilinx/bba/main.cc:269:43: warning: format '%d' expects argument of type 'int', but argument 3 has type 'int64_t' {aka 'long int'} [-Wformat=]
24.66   269 |             printf("    stream '%s' with %d tokens\n", s.name.c_str(), int64_t(s.tokenTypes.size()));
24.66       |                                          ~^                            ~~~~~~~~~~~~~~~~~~~~~~~~~~~~
24.66       |                                           |                            |
24.66       |                                           int                          int64_t {aka long int}
24.66       |                                          %ld
24.66 /nextpnr-xilinx/bba/main.cc:397:28: warning: format '%x' expects argument of type 'unsigned int', but argument 2 has type 'int64_t' {aka 'long int'} [-Wformat=]
24.66   397 |                 printf("%08x ", cursor - numBytes);
24.66       |                         ~~~^    ~~~~~~~~~~~~~~~~~
24.66       |                            |           |
24.66       |                            |           int64_t {aka long int}
24.66       |                            unsigned int
24.66       |                         %08lx
24.66 /nextpnr-xilinx/bba/main.cc:449:42: warning: format '%d' expects argument of type 'int', but argument 4 has type 'int64_t' {aka 'long int'} [-Wformat=]
24.66   449 |         fprintf(fileOut, "const char %s[%d] =\n\"", streams[0].name.c_str(), int64_t(data.size()) + 1);
24.66       |                                         ~^                                   ~~~~~~~~~~~~~~~~~~~~~~~~
24.66       |                                          |                                                        |
24.66       |                                          int                                                      int64_t {aka long int}
24.66       |                                         %ld
```

hansfbaier, 2024-08-28:
Do you build it with nix?

closed 2024-08-28 by hansfbaier

hsk, 2024-08-28:
No, I build it with Docker, see here https://github.com/FPGAOL-CE/osstoolchain-docker-things/tree/master .
But thank you for the quick response.

hsk, 2024-08-28:
After that, there were problems such as insufficient memory, but when make -j8 was changed to make -j4, it was possible to build without problems and even create an image file.
Thank you very much.

regymm, 2024-08-29:
Please try building nextpnr with single thread (-j1). I encountered some problems with multi-thread build. Suspected to be ARM's memory order issue, but has no proof now.

commit 9debb871 (2024-08-28): bba: fix compiler warnings (fixes #42)