openXC7/nextpnr-xilinx #44: Faild to parse newly add data to architecture (chip_info) .bin file

weikongwei, 2024-09-11:
I'm have added code in bbaexport.py to write clock region data to the .bba file, and thereafter, it is converted into .bin file. Then, it can be loaded as memory-mapped-file in Nextpnr.
However, when I try to debug NextPNR, i'm not able to access clock region data by using [] operator defined in struct RelPtr. The compiler tells me "Could not find operator []."

I have verified the offest added to chip_info, no issues found.
I also modified the chipdb.hexpat file to visulize .bin file in ImHex. No ssues found, all data seems correct.
When I modified the .hexpat file, I realized that nextpnr may not have the knowledge to map clock region data from .bin to the struct ClockRegionDataPOD defined in NextPNR.
So far, no clue to complete this mapping.

When writing .bba file,  `bba.ref("clock_regions")` is added to  clock_regions to refer back to the head location of the clock region data like shown below.

**bbaexport.py:**
```
print("Exporting clock region data...")
bba.label("clock_regions")
for clock_region in d.clock_regions:
	bba.u32(seen_clockregion[clock_region.name]) # clock region name, constid
	bba.u32(clock_region.x0) # lower bound of x
	bba.u32(clock_region.y0) # lower bound of y
	bba.u32(clock_region.x1) # upper bound of x
	bba.u32(clock_region.y1) # upper bound of y

# Main chip info structure
bba.label("chip_info")
bba.str(d.name) # device name char*
bba.str("prjxray") # generator name char*
bba.u32(1) # version
bba.u32(d.width) # tile grid width
bba.u32(d.height) # tile grid height
bba.u32(len(d.clock_regions)) # number of clock regions
bba.u32(len(tile_insts)) # number of tiles
bba.u32(len(tile_types)) # number of tiletypes
bba.u32(len(node_wire_count)) # number of nodes
bba.ref("clock_regions") # clock region data
bba.ref("tiletype_data") # reference to tiletype data list
bba.ref("tile_insts") # reference to list of tile instances
bba.ref("nodes") # reference to list of nodes
bba.ref("extra_constids") # reference to list of constid strings (extra to baked-in ones)
bba.u32(1) # only one speed grade currently
bba.ref("timing") # timing data
bba.pop()
```

**xilinx/arch.h:**
```
NPNR_PACKED_STRUCT(struct ClockRegionDataPOD {
    int32_t clock_region_name;  // constid
    int32_t x0, y0;  // lower bounds for bounding box
    int32_t x1, y1;  // upper bounds 
});

NPNR_PACKED_STRUCT(struct ChipInfoPOD {
    RelPtr<char> name;
    RelPtr<char> generator;

    int32_t version;
    int32_t width, height;
    int32_t num_of_clock_regions;
    int32_t num_tiles, num_tiletypes, num_nodes;
    RelPtr<ClockRegionDataPOD> clock_regions;
    RelPtr<TileTypeInfoPOD> tile_types;
    RelPtr<TileInstInfoPOD> tile_insts;
    RelPtr<NodeInfoPOD> nodes;

    RelPtr<ConstIDDataPOD> extra_constids;

    int32_t num_speed_grades;
    RelPtr<TimingDataPOD> timing_data;
});

```

**Chip_info shown in ImHex:**
 [image]

hansfbaier, 2024-09-11:
That might be a problem of the debugger. The C++ support of debuggers is not always great.
Try a cout or printf that should work better.

weikongwei, 2024-09-12:
> That might be a problem of the debugger. The C++ support of debuggers is not always great. Try a cout or printf that should work better.

Thanks a lot! 
It is readable when using std::cout. The problem is the debugger indeed.

closed 2024-09-12 by weikongwei