openXC7/nextpnr-xilinx #50: routing hangs for small 8k design without BRAMs

mirekez, 2024-10-24:
[cant_route_overused_3.zip](https://github.com/user-attachments/files/17505283/cant_route_overused_3.zip)

hansemro, 2024-10-29:
Hi @mirekez , I noticed there was an error during post-placement validation which may have caused routing to fail. Do you mind sharing a post-placement netlist json file as I was unable to replicate with current openxc7 release?

Please run the following and share the generated netlist file `TestMesh.place.json`:
```
# synthesize design
yosys -p "clkbufmap -inpad IBUFG *clock*; synth_xilinx -flatten  -arch xc7 -top TestMesh; write_json TestMesh.json" TestMesh.sv 
# skip route step and generate post-placement netlist
nextpnr-xilinx --chipdb ../../../chipdb//xc7a100tcsg324.bin --xdc TestMesh.xdc --json TestMesh.json --no-route --write TestMesh.place.json
```

hansfbaier, 2026-09-19:
Checked with the current toolchain (toolchain-nix main → nextpnr-xilinx 0.9.5, yosys 0.67+post, fresh `xc7a100tcsg324-2` chipdb built with the current `bbaexport`) against the attachment: the design routes to completion, so this no longer reproduces.

From the attached `stderr.log`: overuse 5927 → 61 → 4 → **3 by iteration 4, then 1093 of its 1096 iterations at exactly `overused=3`** until the run was killed. Note that log already says `Running router2`, so this is the same router that is the default today — it was a stuck conflict, not a superseded algorithm.

Results now, on the attached `TestMesh.json`/`TestMesh.xdc` and the exact command from the report:

| run | result |
|---|---|
| as reported | exit 0, overuse 4963 → 0 in **7 iterations** |
| `--seed 1 / 5 / 42` | exit 0, 0 overused (6 / 7 / 7 iterations) |
| re-synthesized with current yosys (2739 LCs vs 2720 then) | exit 0, 0 overused |
| through to a bitstream | `fasm2frames` + `xc7frames2bit` → 3,825,895 B |

Placement reproduces the attached log almost exactly — same `post-place repair: relocated 1744 stranded cluster(s)/cell(s)` — so what changed is routing: router2 now reports `Adopted 4968 pre-routed arc(s)` (constant/carry-network arcs arrive pre-routed and are inherited rather than re-routed from scratch), plus the router2 convergence work since 0.9.x. No `unable to route`, no overused wires left at the end, timing 79–100 MHz against the 12 MHz target.

Caveat on the claim: verified at the flow level (routing converges, FASM and bitstream are produced). No A7-100T here, so it was not programmed or run on silicon.

Closing as no longer reproducible with 0.9.5. If a design like this still plateaus on a current build, please reopen with the synthesized `.json` and the chipdb revision — a stuck `overused=N` that never moves is exactly the kind of thing that needs the netlist to chase.

closed 2026-09-19 by hansfbaier