openXC7/nextpnr-xilinx #51: json2dcp: replace deprecated RapidWright API; fix cell placement and routing

hansemro, 2024-10-30:
`json2dcp` has not been maintained for some time but could be useful for post-placement/routing validation and DRC in Vivado. While replacing deprecated RapidWright API is fairly straightforward, it will be less trivial to fix json2dcp to handle a post-route netlist json file produced with more recent versions of nextpnr-xilinx.

From what I have experimented so far, the following needs work:
- IOB PAD/IBUF/OBUF cell instantiation/placement and routing
    - [`createAndPlaceIOB`](https://www.rapidwright.io/javadoc/com/xilinx/rapidwright/design/Design.html#createAndPlaceIOB-java.lang.String-com.xilinx.rapidwright.design.PinType-java.lang.String-java.lang.String-) should be used to create IBUF/OBUF cells but require specifying IOSTANDARD and PACKAGE_PIN attributes which are retained in PAD cells.
        - Not sure how to create PAD cells or whether it is necessary to do so.
- inter-site and intra-site net/PIP routing
    - In general, when handling intra-site (`SITEWIRE/<SITE>/<SRCNET>` to `SITEWIRE/<SITE>/<SNKNET>`) routing, we can use [`routeIntraSiteNet`](https://www.rapidwright.io/javadoc/com/xilinx/rapidwright/design/SiteInst.html#routeIntraSiteNet-com.xilinx.rapidwright.design.Net-com.xilinx.rapidwright.device.BELPin-com.xilinx.rapidwright.device.BELPin-)
    - Haven't yet certain how to handle site-to-interconnect and interconnect-to-site routing.
- Enhancement: ingest a post-placement netlist, skip routing processing, and produce DCP with just cell placement.
    - This could be useful for testing routing feasibility using Vivado's router as we test with more congested designs.

Experimental branch: https://github.com/hansemro/nextpnr-xilinx/commits/rapidwright-2024.1.0/
- Tested RapidWright commit: https://github.com/Xilinx/RapidWright/commits/1432ed645b129225f4af1f7111b1fa3d28511e02
- GSON 2.11.0: https://github.com/google/gson/releases/tag/gson-parent-2.11.0
- Currently does not yet handle IOB cells properly so designs with them will fail...
    - json2dcp should be able to process [primitive-tests/clb-tests/jtag-test](https://github.com/openXC7/primitive-tests/tree/main/clb-tests/jtag-test) without debug LEDs.
- To enable json2dcp build, specify the following cmake options: `-DRAPIDWRIGHT_PATH=/path/to/RapidWright -DGSON_PATH=/path/to/gson.jar`

hansemro, 2024-11-01:
Ported json2dcp to python jupyter notebook with routing handling improvements: https://github.com/hansemro/json2dcp-notebook/blob/main/json2dcp.ipynb

This should hopefully help out anyone interested in experimenting with this work.

hansemro, 2024-11-02:
Experimental branch with inter-site/tile net routing: https://github.com/hansemro/json2dcp-notebook/tree/inter-site-routing-dev

Fully routed jtag-test design:
[image]

Now focusing on identifying remaining routing and DRC issues...

hansemro, 2024-11-03:
Currently, [`createIBUFDS`](https://www.rapidwright.io/javadoc/com/xilinx/rapidwright/design/Design.html#createIBUFDS-java.lang.String-java.lang.String-com.xilinx.rapidwright.device.Site-com.xilinx.rapidwright.device.Site-com.xilinx.rapidwright.design.Net-com.xilinx.rapidwright.edif.EDIFNet-java.lang.String-) only supports Ultrascale series with DIFFINBUF+IBUFCTRL BELs, so this will have to be handled ourselves.

hansemro, 2024-11-07:
I am seeing some bogus PIPs in the nextpnr netlist:

- incorrect INT tile prefix for GND_WIRE and VCC_WIRE routing nodes (from TIEOFF sites)
    - Example of invalid PIP: `INT_L_X0Y134/VCC_WIRE->INT_R_X25Y134/FAN_ALT7`
        - Note that the horizontal position is off (and possibly why the prefix between source and sink tiles do not match)
        - Incorrect position may have derived from the following: `NULL_X0Y260/PSEUDO_VCC_WIRE_GLBL->INT_L_X0Y134/VCC_WIRE`
    - json2dcp workaround: use sink tile prefix when retrieving source GND_WIRE/VCC_WIRE routing nodes.
        - https://github.com/hansemro/json2dcp-notebook/commit/a7a449749867ce40497cbd8b8b6565a21b0ffba5
- invalid site pin destination in some CLB tile to site routings:
    - Bogus PIP (D5 != D1): `CLBLM_R_X25Y133/CLBLM_M_D5->SITEWIRE/SLICE_X36Y133/D1`

clb jtag test json netlist: https://gist.github.com/hansemro/dea2b6b9eac13566f113f7a91474b5d1

hansemro, 2024-11-07:
I am seeing some bogus PIPs in the nextpnr netlist:

- incorrect INT tile prefix for GND_WIRE and VCC_WIRE routing nodes (from TIEOFF sites)
    - Example of invalid PIP: `INT_L_X0Y134/VCC_WIRE->INT_R_X25Y134/FAN_ALT7`
        - Note that the horizontal position is off (and possibly why the prefix between source and sink tiles do not match)
        - Incorrect position may have derived from the following: `NULL_X0Y260/PSEUDO_VCC_WIRE_GLBL->INT_L_X0Y134/VCC_WIRE`
    - json2dcp workaround: use sink tile name when retrieving source GND_WIRE/VCC_WIRE routing nodes.
        - https://github.com/hansemro/json2dcp-notebook/commit/a7a449749867ce40497cbd8b8b6565a21b0ffba5
- invalid site pin destination in some CLB tile to site routings:
    - Bogus PIP (D5 != D1): `CLBLM_R_X25Y133/CLBLM_M_D5->SITEWIRE/SLICE_X36Y133/D1`

clb jtag test json netlist: https://gist.github.com/hansemro/dea2b6b9eac13566f113f7a91474b5d1

> Currently, [createIBUFDS](https://www.rapidwright.io/javadoc/com/xilinx/rapidwright/design/Design.html#createIBUFDS-java.lang.String-java.lang.String-com.xilinx.rapidwright.device.Site-com.xilinx.rapidwright.device.Site-com.xilinx.rapidwright.design.Net-com.xilinx.rapidwright.edif.EDIFNet-java.lang.String-) only supports Ultrascale series with DIFFINBUF+IBUFCTRL BELs, so we will need to find a different way to support 7 series.

IBUFDS support for 7 series being experimented in https://gist.github.com/hansemro/24adef209d9ff91f2c01b6f053990861

hansemro, 2024-11-09:
Finally got around to handling IBUFDS in https://github.com/hansemro/json2dcp-notebook/commit/9a41b8296c2e49593eaddbc3cd49eaad1034fa92.

hansfbaier, 2025-03-25:
Where did you find json2dcp? I did not find it in nextpnr-xilinx nor prjxray, nor searching the net gave anything useful.
Did you write it yourself?

hansemro, 2025-03-25:
> Where did you find json2dcp? I did not find it in nextpnr-xilinx nor prjxray, nor searching the net gave anything useful.

Located at nextpnr-xilinx/xilinx/java/json2dcp.java:
- https://github.com/openXC7/nextpnr-xilinx/blob/stable-backports/xilinx/java/json2dcp.java
- https://github.com/gatecat/nextpnr-xilinx/blob/xilinx-upstream/xilinx/java/json2dcp.java

> Did you write it yourself?

The original java utility was left abandoned for some time (without following RapidWright API changes over time) but it seemed promising for checking the netlist visually and for bugs. To make testing changes a bit easier for myself, I ported the project as a python notebook and later made improvements to fix placement and routing bugs that I encountered. There are still some bugs left but some of it comes from nextpnr-xilinx's final json netlist missing some details (missing PIPs/SitePIPs) that make filling in the gaps a bit more challenging (which also raises some questions of the validity of json2dcp as a verification tool).

hansfbaier, 2026-09-24:
Not carried over: **fork-only**. The live engine has no json2dcp/RapidWright/DCP path and does not want one.

The tools this issue maintains are the fork's Java helpers, built only when `RAPIDWRIGHT_PATH`/`GSON_PATH` are set (`nextpnr-xilinx/xilinx/family.cmake:1-22` → `rapidwright_bbaexport` / `rapidwright_json2dcp`). `nextpnr/himbaechel/uarch/xilinx/` has no counterpart, and the port's plan excludes the RapidWright/xcup/DCP flow by decision (`docs/xilinx-porting/05-porting-plan.md:52-55, 230-233`). The `X_ORIG_TYPE` / `X_ORIG_PORT_*` attributes do exist on the live engine, but they are load-bearing for its FASM writer (`fasm.cc:554, 599-603, 623, 769, 964, 1675, 1703, 2213`), not evidence of an exporter requirement.

Closing with the archive; the code and this thread stay readable.

closed 2026-09-24 by hansfbaier