openXC7/nextpnr-xilinx #57: OSERDESE2 SHIFTIN/OUT breaks DDR3 controller

regymm, 2024-12-08:
The latest stable-backports breaks UberDDR3 by this error:

```
ERROR: OSERDESE2 'ddr3_top_inst.ddr3_phy_inst.genblk1[0].OSERDESE2_cmd' can only be connected to the SHIFOUT port of another slave OSERDESE2, but is connected to this master: '$PACKER_GND_DRV'
```

It seems the assumptions about SHIFTIN/SHIFTOUT are not very suitable [here](https://github.com/openXC7/nextpnr-xilinx/blob/stable-backports/xilinx/pack_io_xc7.cc#L855)? 
By my naive understanding, the [DDR3 controller](https://github.com/AngeloJacobo/UberDDR3/blob/main/rtl/ddr3_phy.v#L211) doesn't use this function, so it ties SHIFTIN to zero and SHIFTOUT unconnected. It works on Vivado without warning. 

CC @AngeloJacobo

hansfbaier, 2024-12-08:
@regymm Which toolchain/branch of nextpnr-xilinx are you using?
The current stable-backports which is used in the nix toolchain has code to disconnect the SHIFTIN ports if
they are wired to ground.
[image]

regymm, 2024-12-09:
I'm using this repo's stable-backports branch, compiled from source code. Also tried the nix version and it seems the same. 
run `make` in Nix, in e.g. `UberDDR3/example_demo/arty_s7` can repeat the issue.

AngeloJacobo, 2025-02-22:
Hi @hansfbaier @regymm , commenting out SHIFTIN1/SHIFTIN2 fixes the problem on the PHY of [UberDDR3](https://github.com/AngeloJacobo/UberDDR3). OpenXC7 can now finish the run. It seems connecting SHIFTIN1/SHIFTIN2 to zero causes the issue, it must be explicitly disconnected.

regymm, 2025-02-22:
This issue https://github.com/chili-chips-ba/openeye-CamSI/issues/38 may be closely related. It's also caused by connecting I/OSERDESE2 pins (D) to zero instead of floating.

hansfbaier, 2025-02-23:
@regymm, @AngeloJacobo I pushed a fix, which seemed to fix it for me, please confirm:
https://github.com/openXC7/nextpnr-xilinx/commit/ed0cf8a2070fa8733eee651f85ac643519cf32af

AngeloJacobo, 2025-03-02:
Hi @hansfbaier , the run ends with the same error:
`ERROR: OSERDESE2 'ddr3_top.ddr3_phy_inst.genblk1[0].OSERDESE2_cmd' can only be connected to the SHIFTOUT port of another slave OSERDESE2, but is connected to this master: '$PACKER_GND_DRV'`

I already re-run the command:
` $ nix develop github:openxc7/toolchain-nix`

This happens when I run make for the arty-s7 demo project (openxc7_run branch):
https://github.com/AngeloJacobo/UberDDR3/tree/openxc7_run/example_demo/arty_s7

regymm, 2025-03-02:
I was working on another branch and this seems solved (on the openeye-CamSI project). Maybe you can try using `nix develop --refresh github:openxc7/toolchain-nix` to force update the nix toolchain?

AngeloJacobo, 2025-03-02:
I tried `nix develop --refresh github:openxc7/toolchain-nix`, but I'm still getting the same error when running openxc7 here: https://github.com/AngeloJacobo/UberDDR3/tree/openxc7_run/example_demo/arty_s7. I did remember that when I did the first nix develop command today, it took some time to download something so I thought I already had the latest version.

hansfbaier, 2025-03-02:
@AngeloJacobo Can you try and rebuild your project after a make clean?
I get a yosys build error when trying to build your design, so I cannot reproduce the PnR issue:
```
...
8.4.2. Executing AST frontend in derive mode using pre-parsed AST for module `\ddr3_top'.
Parameter \CONTROLLER_CLK_PERIOD = 12000
Parameter \DDR3_CLK_PERIOD = 3000
Parameter \ROW_BITS = 14
Parameter \COL_BITS = 10
Parameter \BA_BITS = 3
Parameter \BYTE_LANES = 2
Parameter \AUX_WIDTH = 16
Parameter \WB2_ADDR_BITS = 32
Parameter \WB2_DATA_BITS = 32
Parameter \MICRON_SIM = 0
Parameter \ODELAY_SUPPORTED = 0
Parameter \SECOND_WISHBONE = 0
Parameter \WB_ERROR = 0
Parameter \ECC_ENABLE = 0
Generating RTLIL representation for module `$paramod$0e71a046ea92df5cd086e9a531a2edcbd49d252e\ddr3_top'.
ERROR: Module `\xadc_wiz_0' referenced in module `\arty_ddr3' in cell `\xadc_inst' is not part of the design.
make: *** [Makefile:42: arty_ddr3.json] Error 1
```

AngeloJacobo, 2025-03-09:
Hi @hansfbaier , I updated the main repo now, please try running make inside:
https://github.com/AngeloJacobo/UberDDR3/tree/main/example_demo/arty_s7

```
Info: Preparing clocking...
Info: Packing constants..
ERROR: OSERDESE2 'ddr3_top.ddr3_phy_inst.genblk1[0].OSERDESE2_cmd' can only be connected to the SHIFTOUT port of another slave OSERDESE2, but is connected to this master: '$PACKER_GND_DRV'
5 warnings, 1 error
make: *** [Makefile:52: arty_ddr3.fasm] Error 255
```

hansfbaier, 2025-03-10:
@AngeloJacobo I fixed the bugs, and the bitstream builds fine now here, please test:
https://github.com/openXC7/nextpnr-xilinx/commits/stable-backports/

hansfbaier, 2025-03-15:
Yes with nix develop .