openXC7/nextpnr-xilinx #62: openXC7 - ERROR: Failed to route arc 164 of net 'csi_byte_clk', from SITEWIRE/BUFGCTRL_X0Y23/O to SITEWIRE/SLICE_X3Y114/AFFMUX_OUT

harunkovacevic, 2025-02-25:
The issue is explained in this [issue](https://github.com/chili-chips-ba/openeye-CamSI/issues/40) in more detail.

regymm, 2025-04-03:
This commend  https://github.com/chili-chips-ba/openeye-CamSI/issues/40#issuecomment-2692053459 contains a minimum demo of this issue, aka clock input to FDRE
```verilog
`timescale 1ns / 1ps
 
module top(
	input CLK100MHZ,
	output [15:0]LED,
	input [15:0]sw
    );
	reg [31:0]cnt = 0;
	always @ (posedge CLK100MHZ) begin
		cnt <= cnt + 1;
	end
	assign LED[7:0] = cnt[31:24];
	assign LED[15:8] = sw[15:8] & clkr;
    //reg clkr = 0;
    //always @ (posedge CLK100MHZ) begin
        //clkr <= CLK100MHZ;
    //end
    wire clkr;
    FDRE ff (
        .C(CLK100MHZ),
        .CE(1),
        .D(CLK100MHZ),
        .Q(clkr),
        .R(0)
    );
endmodule
```

Name of this issue can be changed to "Failed to route clock when the clock is fed to FDRE D input"

hansfbaier, 2025-04-03:
IIRC you just can't feed clock into D inputs. You need a primitive for that but I forgot which one. Vivado probably infers that. Look into the implementation viewer to confirm.

hansfbaier, 2026-09-19:
Verified fixed — closing.

The reported routing failure no longer occurs on current `main` (`d486d8ef`). The CamSI RTL itself is not in this repo, so I reproduced the minimal demo from the thread — a BUFG clock net feeding an FDRE `D` input, the same topology as the `BUFGCTRL → AFFMUX_OUT` arc in the title — on `xc7a35tcpg236-1` (Basys3):

    yosys -p "read_verilog top.v; synth_xilinx -flatten -abc9 -arch xc7 -top top; write_json top.json"
    nextpnr-xilinx --chipdb xc7a35tcpg236.bin --xdc top.xdc --json top.json --fasm top.fasm

Result: exit 0; `Constrained BUFGCTRL … to bel 'BUFGCTRL_X0Y0/BUFGCTRL'`; `Router2 time 0.66s`; FASM written (md5 `d506f091`); no `Failed to route arc … AFFMUX_OUT`. The netlist exercises the topology — the FF's `D` and `C` are both on the BUFG output net.

Caveat: the title's coordinates (`BUFGCTRL_X0Y23` → `SLICE_X3Y114`) come from the larger CamSI design, which is not available here, so this is the same topology rather than the identical netlist. I did not isolate the single fixing commit (`df858b1e`, *router2: make wire reservation conflict-aware*, is a plausible candidate but I have not verified that attribution). If the full CamSI design still fails, please reopen and I will dig in.

closed 2026-09-19 by hansfbaier