openXC7/nextpnr-xilinx #65: Unsupported clock edge parameter SAME_EDGE_PIPELINED

janrinze, 2025-03-02:
I recently changed from https://github.com/gatecat/nextpnr-xilinx to this version of nextpnr-xilinx.

The module IDDR has 3 different settings (see : https://docs.amd.com/r/en-US/ug953-vivado-7series-libraries/IDDR )

With the version from Gatecat I can choose from all three. But with the recently build version from here I get:

ERROR: unsupported clock edge parameter for cell 'Rx_4564d718_0323_4abb_8e9f_cf0e741b3fb3.IDDR_inst_clk' at ILOGIC_X1Y83: SAME_EDGE_PIPELINED. Supported are: SAME_EDGE and OPPOSITE_EDGE

Using grep I could find that these are defined in Proj X-Ray.

The code used:
```
wire Rx_1,Rx_2;

IDDR #( 
 .DDR_CLK_EDGE("SAME_EDGE_PIPELINED"), // "OPPOSITE_EDGE", "SAME_EDGE" or "SAME_EDGE_PIPELINED"
 .INIT_Q1(1'b0),  // Initial value of Q1: 1'b0 or 1'b1
 .INIT_Q2(1'b0),  // Initial value of Q2: 1'b0 or 1'b1
 .SRTYPE("SYNC")  // Set/Reset type: "SYNC" or "ASYNC"
) IDDR_inst_clk (
   .Q1(Rx_1), // 1-bit output for positive edge of clock
   .Q2(Rx_2), // 1-bit output for negative edge of clock
   .C(clk),     // 1-bit clock input
   .CE(1'd1),     // 1-bit clock enable input
   .D(Rx),  // 1-bit DDR data input
   .R(1'd0),      // 1-bit reset
   .S(1'd0)       // 1-bit set
   );
```

hansfbaier, 2025-03-02:
Hi, I cannot find direct support for SAME_EDGE_PIPELINED in prjxray-db, except for this feature:
```
$ rg PIPELINED .
./artix7/timings/RIOI3_SING.sdf
142:        (CELLTYPE "ILOGICE3_IFF_PIPELINED")

./artix7/timings/RIOI3_TBYTETERM.sdf
142:        (CELLTYPE "ILOGICE3_IFF_PIPELINED")

./artix7/timings/RIOI3_TBYTESRC.sdf
142:        (CELLTYPE "ILOGICE3_IFF_PIPELINED")

./artix7/timings/LIOI3.sdf
142:        (CELLTYPE "ILOGICE3_IFF_PIPELINED")

./artix7/timings/LIOI3_TBYTETERM.sdf
142:        (CELLTYPE "ILOGICE3_IFF_PIPELINED")

./artix7/timings/LIOI3_SING.sdf
142:        (CELLTYPE "ILOGICE3_IFF_PIPELINED")

./artix7/timings/RIOI3.sdf
142:        (CELLTYPE "ILOGICE3_IFF_PIPELINED")

./spartan7/timings/RIOI3_SING.sdf
142:        (CELLTYPE "ILOGICE3_IFF_PIPELINED")

./artix7/timings/LIOI3_TBYTESRC.sdf
142:        (CELLTYPE "ILOGICE3_IFF_PIPELINED")

./spartan7/timings/RIOI3_TBYTETERM.sdf
142:        (CELLTYPE "ILOGICE3_IFF_PIPELINED")

./spartan7/timings/LIOI3_SING.sdf
142:        (CELLTYPE "ILOGICE3_IFF_PIPELINED")

./spartan7/timings/RIOI3_TBYTESRC.sdf
142:        (CELLTYPE "ILOGICE3_IFF_PIPELINED")

./spartan7/timings/RIOI3.sdf
142:        (CELLTYPE "ILOGICE3_IFF_PIPELINED")

./spartan7/timings/LIOI3.sdf
142:        (CELLTYPE "ILOGICE3_IFF_PIPELINED")

./spartan7/timings/LIOI3_TBYTESRC.sdf
142:        (CELLTYPE "ILOGICE3_IFF_PIPELINED")

./spartan7/timings/LIOI3_TBYTETERM.sdf
142:        (CELLTYPE "ILOGICE3_IFF_PIPELINED")

./zynq7/timings/RIOI3_SING.sdf
142:        (CELLTYPE "ILOGICE3_IFF_PIPELINED")

./zynq7/timings/RIOI3_TBYTETERM.sdf
142:        (CELLTYPE "ILOGICE3_IFF_PIPELINED")

./zynq7/timings/RIOI3_TBYTESRC.sdf
142:        (CELLTYPE "ILOGICE3_IFF_PIPELINED")

./zynq7/timings/LIOI3.sdf
142:        (CELLTYPE "ILOGICE3_IFF_PIPELINED")

./zynq7/timings/LIOI3_TBYTETERM.sdf
142:        (CELLTYPE "ILOGICE3_IFF_PIPELINED")

./zynq7/timings/LIOI3_SING.sdf
142:        (CELLTYPE "ILOGICE3_IFF_PIPELINED")

./zynq7/timings/RIOI3.sdf
142:        (CELLTYPE "ILOGICE3_IFF_PIPELINED")

./zynq7/timings/LIOI3_TBYTESRC.sdf
142:        (CELLTYPE "ILOGICE3_IFF_PIPELINED")
```
Can you submit a complete minimal example design that uses that feature,
together with a vivado project file.
Then we could disassemble the bitstream and see what features are used, 
and if there are unknown bits.

hansfbaier, 2025-03-02:
Hi, it looks like the above feature is the one.
I pushed an implementation here:
https://github.com/openXC7/nextpnr-xilinx/tree/same-edge-pipelined
Please test it and let me know if it works.

janrinze, 2025-03-04:
Hi, I need to check a bit more with the scope to verify since the difference is only one clock on the inputs.
I'll get back here soon to update the status.

hansfbaier, 2025-03-07:
@janrinze If you send me a bitstream of a test design with SAME_EDGE_PIPELINES compiled with Vivado, together with the part number, then I can disassemble it and see if that feature is set.

hansfbaier, 2025-03-24:
@janrinze Have you been able to test it? Then we can merge it soon