openXC7/nextpnr-xilinx #68: Out of range error

janrinze, 2025-07-13:
The example builds for ecp5 platforms but fails for xc7.
I'm sure I must be doing something wrong here.

example : [Out of range error](https://github.com/janrinze/Xilinx_Tests/tree/main/out_of_range)

janrinze, 2025-08-07:
Today got this ridiculous message:
terminate called after throwing an instance of 'std::out_of_range'
  what():  vector::_M_range_check: __n (which is 328) >= this->size() (which is 0)

So we have a 328 bit value that does not really exists..
Even this error should have at least a sensible context as to what this 328 bit thingy refers to.
It's becoming a real issue lately.

janrinze, 2025-08-07:
Using gdb I found the culprit:
nextpnr_xilinx::(anonymous namespace)::GenericFrontend<nextpnr_xilinx::JsonFrontendImpl>::merge_nets

This is in the source code at nextpnr-openxc7/frontend/frontend_base.h:691

Apparently at yosysHQ there are many patches for the frontend after 2020 which never made it into nextpnr-xilinx.

Not sure if I should do work on this myself or if someone more knowledgeable is available to 'upgrade' the source code.

hansfbaier, 2026-01-17:
@janrinze Thanks, great find, do you want to make a PR for the stable-backports branch?

hansfbaier, 2026-09-19:
Verified fixed — closing.

Confirmed with a pre-fix A/B, not merely "it works now". I built nextpnr-xilinx at `c7c655d5^` (`72d8217b`, self-built, with a chipdb written by that revision's own `bbaexport.py`/`bbasm`) and ran the example from @janrinze's `Xilinx_Tests/out_of_range` (xc7a200t):

* **pre-fix**: exit `134` — `terminate called after throwing an instance of 'std::out_of_range'` / `what(): vector::_M_range_check: __n (which is 32) >= this->size() (which is 0)`, backtrace frame in `nextpnr_xilinx::(anonymous namespace)::GenericFrontend<nextpnr_xilinx::JsonFrontendImpl>::merge_nets` at `frontend/frontend_base.h:694` — the exact function found with gdb in the report above.
* **current** (0.9.6 and current `main`): exit 0, FASM produced, `Max frequency for clock clk_o: 258.67 MHz (PASS at 12.00 MHz)`, no `out_of_range` in any log.

The out-of-range index differs from the original report (`32` vs `328`) because the netlist comes from a different yosys build; the root cause is the same — `merge_nets` does `net_old_indices.at(net->udata)` for a net whose entry is missing because `net_old_indices` was not kept in lockstep with `net_flatindex`.

**The fix is `c7c655d5`** — *frontend: keep net merge history for imported nets* (2026-03-12), which adds the two missing entries (`net_old_indices.emplace_back()` after each `net_flatindex.push_back`). It is an ancestor of current `main`. Checked with `git patch-id --stable`, it is **byte-identical to the upstream patch linked in the thread**, YosysHQ/nextpnr `af7be215` (*Improve handling of unflattened netlists*, 2020-03-06): both yield patch-id `8577066bc6f67634c1f25e242bca39acdb0fc5d3`. Upstream's commit itself is still not in this history — the openXC7 re-application of it is what closed this.

Also worth recording for anyone who hits this shape of crash: the failure is deterministic and happens during netlist import, before placement — so it is reproducible in under a second and independent of seeds, threading or load.

Thanks for the pointer to the upstream patch; it turned this into a quick confirmation.

closed 2026-09-19 by hansfbaier